dw-0/kiauh #416: Bad substitution -> Syntax error

catroc, 2023-12-23:
### Linux Distribution

Raspi 4 - bookworm

### What happened

Just want to start KIAUH ...

Gives me:
```
kiauh.sh: 16: Bad substitution
kiauh.sh: 14: -/scripts/backup.sh: Syntax error: "(" unexspected
```

### What did you expect to happen

Well ... starting KIAUH ^^

### How to reproduce

`sh ./kiauh/kiauh.sh`

### Additional information

I have installed it many times, tried an new image and a new sd card ...
Same problem when I install over git or copy it with winscp.

pardovot, 2024-01-21:
Same issue, after each time I restart the system or something this goes corrupted or something.

itsj, 2024-01-27:
Same issue for me. Installation via git.

itsj, 2024-01-27:
I've found the solution. You have to start the script from your home directory with
./kiauh/kiauh.sh
Before I start it within the kiauh directory.
Hope that help.

dw-0, 2024-01-27:
Im unable to reproduce the issue on Debian 12 Bookworm and Ubtunu 22.04.3 LTS:
[image]
[image]

dw-0, 2024-01-27:
> I found a solution that works for me. You need to start the script from your home directory ./kiauh/kiauh.sh Before that I launched it in the Kiauh directory. Hope that helps.

That shouldn't make a difference.
`kiauh/kiauh.sh` does work as well as `cd kiauh` and then running `./kiauh.sh`

itsj, 2024-01-27:
Ahh. I try to start it with
sh kiauh.sh
So, my fault.

dw-0, 2024-01-27:
> ### How to reproduce
> `sh ./kiauh/kiauh.sh`

@catroc why do you execute it with sh instead of bash? That's your issue...

itsj, 2024-01-27:
It could be that @catroc tried following the same instructions as me.

dw-0, 2024-01-27:
Where do those instructions come from? @itsj

pardovot, 2024-01-27:
I probably did the same mistake, didn't came from any instructions, just the power of habit.

itsj, 2024-01-27:
Sorry, I can't retrace that anymore, too many videos. A German YouTuber.

dw-0, 2024-01-27:
Okay, no problem. I think we can close this issue then. It seems that everything boils down to executing the script in an incorrect way :)

closed 2024-01-27 by dw-0

itsj, 2024-01-27:
Your right, now it works like a charm!