dw-0/kiauh #419: Unable to open file

shodanx2, 2024-01-03:
### Linux Distribution

debian 12.1

### What happened

Trying to print, returns error "Unable to open file"

### What did you expect to happen

It should print the file

### How to reproduce

install debian
follow instructions for 4 instances
connect one kp3s flashed with klipper
press print

### Additional information

Hi,

Here is my dashboard with some more context

[image]

Despite putting in my printer.cfg

    [virtual_sdcard]
    path: /home/user/printer_1_data/gcodes

as requested by moonraker's log

I still get the error message

    Received: '/home/user/printer_data/gcodes'
    Expected: '/home/user/printer_1_data/gcodes'
    
    Modify the [virtual_sdcard] section Klipper's configuration to correct this error.

I tried editing mainsail.cfg, but it is read-only (and I assume, auto generated so that my changes would get overwritten anyway ?)

Also I see I have many such "mainsail.cfg files on my system

    user@klipper:~$ sudo find / | grep mainsail.cfg
    /home/user/printer_3_data/config/mainsail.cfg
    /home/user/mainsail-config/mainsail.cfg
    /home/user/printer_2_data/config/mainsail.cfg
    /home/user/printer_1_data/config/mainsail.cfg
    /home/user/printer_4_data/config/mainsail.cfg

And here is what they contain
    
    user@klipper:~$ grep -A 1 -i "virtual" `sudo find / | grep mainsail.cfg`
    /home/user/printer_3_data/config/mainsail.cfg:[virtual_sdcard]
    /home/user/printer_3_data/config/mainsail.cfg-path: ~/printer_data/gcodes
    --
    /home/user/mainsail-config/mainsail.cfg:[virtual_sdcard]
    /home/user/mainsail-config/mainsail.cfg-path: ~/printer_data/gcodes
    --
    /home/user/printer_2_data/config/mainsail.cfg:[virtual_sdcard]
    /home/user/printer_2_data/config/mainsail.cfg-path: ~/printer_data/gcodes
    --
    /home/user/printer_1_data/config/mainsail.cfg:[virtual_sdcard]
    /home/user/printer_1_data/config/mainsail.cfg-path: ~/printer_data/gcodes
    --
    /home/user/printer_4_data/config/mainsail.cfg:[virtual_sdcard]
    /home/user/printer_4_data/config/mainsail.cfg-path: ~/printer_data/gcodes

Should I be editing all these files manually when i want to change settings ?

Or is there a easy fix I'm not finding here ?

Note this install on x86 was done using [kiauh](https://github.com/dw-0/kiauh)

This seems to be a recurrent speed bump to new users.

    https://old.reddit.com/r/klippers/comments/zyfmlz/gcode_path_received_from_klipper_does_not_match/
    https://old.reddit.com/r/klippers/comments/zp55sp/gcode_path_received_from_klipper_does_not_match/
    https://old.reddit.com/r/klippers/comments/13pcrsy/unable_to_upload_gcode_files_larger_than_20mb_to/
    https://old.reddit.com/r/klippers/comments/z75269/klipper_not_flashing_and_mainsail_giving_errors/
    https://old.reddit.com/r/klippers/comments/10g498w/in_mainsail_how_do_i_add_multiple_printers/
    https://klipper.discourse.group/t/gcode-path-received-from-klipper-does-not-match-expected-path/4513
    https://github.com/Arksine/moonraker/issues/582
    https://old.reddit.com/r/klippers/comments/y6o05y/unable_to_open_file/
    https://old.reddit.com/r/klippers/comments/10cgle4/unable_to_open_file_error_in_mainsail/
    https://old.reddit.com/r/klippers/comments/zo8a89/unable_to_open_file/
    https://old.reddit.com/r/klippers/comments/o0c0dd/klipper_unable_to_open_gcode/
    https://old.reddit.com/r/klippers/comments/zf616o/mainsail_gcode_file_wont_print_unable_to_open_file/
    https://old.reddit.com/r/klippers/comments/zdtosu/hey_i_am_getting_the_klipper_unable_to_open_file/
    https://old.reddit.com/r/klippers/comments/1732ykh/unable_to_open_file/
    https://old.reddit.com/r/klippers/comments/10gditr/fix_for_mainsail_unable_to_open_file_on_fresh/
    https://old.reddit.com/r/klippers/comments/156o1dl/hlep_unable_to_open_file_mainsail/
    https://old.reddit.com/r/klippers/comments/17tort3/unable_to_open_file/
    https://old.reddit.com/r/klippers/comments/189qv1e/klipper_unable_to_open_file/
    https://old.reddit.com/r/klippers/comments/17o36lm/unable_to_open_file_where_to_insert_virtual_sdcard/
    https://old.reddit.com/r/klippers/comments/18es0fw/klipper_unable_to_open_gcode/
    https://old.reddit.com/r/klippers/comments/1156oj2/unable_to_open_file/
    https://old.reddit.com/r/CrealitySonicPad/comments/10rlbmy/key121_unable_to_open_file_gcode/
    https://old.reddit.com/r/klippers/comments/yidpk3/i_can_access_mainsail_remotely_but_when_i_upload/
    https://old.reddit.com/r/klippers/comments/137d8w0/multiple_instances_of_klipper_cant_open_gcode/
    https://old.reddit.com/r/klippers/comments/nsj881/unable_to_upload_gcode_file_within_mainsail/
    https://old.reddit.com/r/klippers/comments/14y7jbk/help_error_unable_to_open_file_on_mainsail/
    https://old.reddit.com/r/klippers/comments/z5z5ds/cant_open_gcodes_due_to_klipper_and_moonraker/
    https://old.reddit.com/r/klippers/comments/10124vv/help_mainsailcfg_in_read_only_mode_cant_edit_it/
    https://old.reddit.com/r/klippers/comments/11nru0d/gcode_path_received_from_klipper_does_not_match/

[image]

more virtual_sd in more files, this is in printer.cfg

    user@klipper:~$ sudo grep -A 1 -i "virtual_sd" `sudo find / | grep printer.cfg`
    /root/kiauh/resources/example.printer.cfg:[virtual_sdcard]
    /root/kiauh/resources/example.printer.cfg-path: %GCODES_DIR%
    --
    /home/user/moonraker/tests/assets/klipper/error_printer.cfg:[virtual_sdcard]
    /home/user/moonraker/tests/assets/klipper/error_printer.cfg-path: ${gcode_path}
    --
    /home/user/moonraker/tests/assets/klipper/base_printer.cfg:[virtual_sdcard]
    /home/user/moonraker/tests/assets/klipper/base_printer.cfg-path: ${gcode_path}
    --
    /home/user/printer_3_data/config/printer.cfg:[virtual_sdcard]
    /home/user/printer_3_data/config/printer.cfg-path: /home/user/printer_3_data/gcodes
    --
    /home/user/printer_2_data/config/printer.cfg:[virtual_sdcard]
    /home/user/printer_2_data/config/printer.cfg-path: /home/user/printer_2_data/gcodes
    --
    /home/user/kiauh/resources/example.printer.cfg:[virtual_sdcard]
    /home/user/kiauh/resources/example.printer.cfg-path: %GCODES_DIR%
    --
    /home/user/printer_1_data/config/printer.cfg:[virtual_sdcard]
    /home/user/printer_1_data/config/printer.cfg-path: /home/user/printer_1_data/gcodes
    --
    /home/user/printer_4_data/config/printer.cfg:[virtual_sdcard]
    /home/user/printer_4_data/config/printer.cfg-path: /home/user/printer_4_data/gcodes

dw-0, 2024-01-03:
This is not really a KIAUH bug, but i will try to help anyways. Otherwise, also checkout the official Klipper discord, there is a good place to get further support.

> Despite putting in my printer.cfg
> 
> ```
> [virtual_sdcard]
> path: /home/user/printer_1_data/gcodes
> ```

Where in your printer.cfg did you put this? It needs to be below the include of the mainsail.cfg. Currently the mainsail.cfg (it's a symlink to another cfg) includes this config block, hence why it is incorrect for pretty much all multi instance installs.
But you can overwrite it by putting another config block in the printer.cfg. It is important that its BELOW the include of the mainsail.cfg.

shodanx2, 2024-01-03:
Hello !

I have originally posted this as a question on the klippers subreddit but only got the suggestion of having edited the wrong printer.cfg and that I should ignore mainsail.cfg.

As I have documented, this appears to be a common issue but is perhaps 'nobody's bug" ? 

As for where in printer.cfg did I put the path, it is there

[image]

Which is above the mainsail.cfg, I have now moved this reference

[image]

After a restart, it appears the error has now cleared

[image]

This is the origin of my printer.cfg file

https://github.com/nehilo/Klipper-KingRoon-Printers/blob/main/KP3S_Configuration/printer.cfg

so this was wrong when I first tried it

```
[virtual_sdcard]
path: ~/gcode_files
```

Since mainsail.cfg appears to overwrite printer.cfg, why are the mainsail.cfg files pointing to ~/printer_data/gcodes instead of /home/user/printer_1_data/gcodes ? Should not that be adjusted during the creation of multiple instances by kiauth ?

In any case, my issue is fixed, now I can move on to the next error message thanks !

dw-0, 2024-01-03:
> Since mainsail.cfg appears to overwrite printer.cfg, why are the mainsail.cfg files pointing to ~/printer_data/gcodes instead of /home/user/printer_1_data/gcodes ? Should not that be adjusted during the creation of multiple instances by kiauth ?

No, it is not intended to be adjusted. The mainsail.cfg comes with it's own repository. During the installation of the macros, while installing Mainsail itself, there is another repository cloned which is used by the mainsail-crew to maintain those macros, that means you can get updated by pulling that repo. I CANNOT modify anything in there, as this would 
1) break the update process
2) require a different modification for each instance
The important file here is the "client.cfg" in the mainsail-config directory. That gets symlinked into each instances config folder. thats why you have so many for them. each printer gets its own symlink. like each printer requires its own printer.cfg.
The only solution and workaround is, to overwrite the parts which are wrong for multi instance setups with the mainsail.cfg.

shodanx2, 2024-01-08:
Ok, well not my setup works.

Figuring this out turned out to be a two weeks speed bump.

In the end, just moving [virtual_sdcard] after include mainsail.cfg in printer.cfg did the trick

I also needed to change the original folder /home/user/printer_1_data/gcodes

I'm starting to think, [virtual_sdcard] simply should not be in printer.cfg ? Since that is not a printer setting, more of something that exists between mainsail and moonraker and maybe klippy ?

Considering that everything else worked out of the box EXCEPT this (and the very old printer.cfg that I got)

I feel like the initial user experience could be greatly improved by getting the various parties to this issue to collaborate instead of letting the user figure it out.

I imagine this is only an issue for people that run multiple printers off a single computer ? And that multiple instance users are not assumed to be first klipper users (like me), so that's probably fewer people.

Anyway, hopefully anyone else with the same error messages will find this conversation and be quickly on their way to printing stuff instead of editing text files !

dw-0, 2024-01-08:
> I imagine this is only an issue for people that run multiple printers off a single computer ? And that multiple instance users are not assumed to be first klipper users (like me), so that's probably fewer people.

Correct. And unless there is a common ground, its required to work around this. Sorry.

closed 2024-01-08 by dw-0

shodanx2, 2024-01-08:
There is a point in the script where it asks you how many instance you will
be running. This interactive part cannot be avoided by anyone. I think it
would be the perfect place to inform the user they will need to do
something about virtual sdcard directive if they ask for more than 1
instance.

> I imagine this is only an issue for people that run multiple printers off
a single computer ? And that multiple instance users are not assumed to be
first klipper users (like me), so that's probably fewer people.
>
> Correct. And unless there is a common ground, its required to work around
this. Sorry.
https://ci6.googleusercontent.com/proxy/zYV_fCRsAAXS4waeB-pBu7KdH_DIKLfyWr-7btSfj6csNvWLhYWIeZOoKYheCgXJXGCvGMLVfckbwVfpuiyvKg-8BmszS-xAdhW6dW-Rs66_4-2oSxstS8RTZ6IzSSvmG_xs6SDLmF3lWt49KkgOo0Vnm-Q5xjiFPXqsGJyuNM1pgR6jQcnuOIOR5z0uKZo-jMGmfqVl9tiVFb0kuFaeRLG8wLLRJn-VEu0xUssIg8lr4wonaQ=s0-d-e1-ft#https://github.com/notifications/beacon/ACRBHPJBEBIOEAATIBO3DZTYNPNRXA5CNFSM6AAAAABBLFHMA6WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTQDO5FC.gif>Message
ID: ***@***.***>

nagynoe13, 2024-03-18:
> Ok, well not my setup works.
>
> Figuring this out turned out to be a two weeks speed bump.
>
> In the end, just moving [virtual_sdcard] after include mainsail.cfg in printer.cfg did the trick
>
> I also [...]

God bless you for this.