dw-0/kiauh #421: Port klippy to STM32MH563ZI

kiminox, 2024-01-07:
### Is your feature request related to a problem? Please describe

Hello,
I am student and I have a project to port klippy/Klipper MCU to STM32MH563ZI.
We use UP4000 for our Klipper Host: this part is installed with kiauh (thanks to them!).
https://github.com/th33xitus/kiauh.git
I saw that we can already build and flash from Klipper Host/KIAUTH for STM32H723, STM32H743 and STM32H750
My question is: do you plan to port klippy to STM32MH563ZI?
If not, can you provide help/advice to develop custom code to port klippy to STM32MH563ZI?
I have no idea where to start?
Thank you

### Describe the solution you'd like

Port klippy to STM32MH563ZI?

dw-0, 2024-01-07:
I think you've got something fundamentally wrong here. KIAUH is a script that only helps to install Klipper (and various other components).

closed 2024-01-07 by dw-0

kiminox, 2024-01-07:
but I saw that we can build and flash from Klipper Host/KIAUTH for STM32H723, STM32H743 and STM32H750?
The code built may be already available, that's what you meant?
In this case, you are right and I close the request then

dw-0, 2024-01-07:
Everything you do in terms of building Klipper is provided by Klipper itself. KIAUH just guides and serves an "Interface" so users are not required to manually execute the commands in the CLI.

Creo2005, 2024-01-07:
> ### Чи пов’язаний ваш запит щодо функції з проблемою? Будь ласка, опишіть
> Привіт, я студент і маю проект портувати klippy/Klipper MCU на STM32MH563ZI. Ми використовуємо UP4000 для нашого Klipper Host: ця частина встановлена ​​за допомогою kiauh (спасибі їм!). https://github.com/th33xitus/kiauh.git Я бачив, що ми вже можемо створювати та флешувати з Klipper Host/KIAUTH для STM32H723, STM32H743 та STM32H750. Моє запитання: чи плануєте ви перенести klippy на STM32MH563ZI? Якщо ні, чи можете ви надати допомогу/пораду щодо розробки спеціального коду для перенесення klippy на STM32MH563ZI? Я не знаю, з чого почати? Дякую
> 
> ### Опишіть рішення, яке ви бажаєте
> Порт klippy до STM32MH563ZI?
> 
> ### Опишіть альтернативи, які ви розглядали
> _Немає відповіді_
> 
> ### Додаткова інформація
> _Немає відповіді_

you have to ask here https://github.com/Klipper3d/klipper

kiminox, 2024-01-07:
ok thank you guys !