dw-0/kiauh #422: KIAUH menu to set raspberry static ip

Nekrofago, 2024-01-17:
### Is your feature request related to a problem? Please describe

Depending on some personal preferences or even some router configurations, some users like me want to set a static IP on the Raspberry Pi after setting up a fresh OS on our SD/USB. It would be awesome to be able to do it via KIAUH

### Describe the solution you'd like

Be able to configure a static IP with the current values or with custom ones and store it in /etc/dhcpcd.conf. It should start asking for the kind of interface to use (eth0, wlan0)

For the current ones, it should store the current local IP, router IP and DNS
For the custom ones, KIAUH should ask for them and probably apply some kind of regex to ensure a correct input.

An example of the result to store:

interface wlan0
static ip_address=XXX.XXX.XXX.XXX/24
static routers=XXX.XXX.XXX.XXX
static domain_name_servers=XXX.XXX.XXX.XXX

### Additional information

I haven't tested it myself, but probably having more than one interface and IP configurations enabled could cause conflicts. Surely, more than one interface defined for the same interface will cause problems.

Then, as we cannot be sure that we are dealing with a fresh install, it should check if there's an already enabled interface and ask to overwrite it

jimmysdiy, 2024-01-26:
KIAUH is for the Klipper Install Aid.. Your ask isnt bad, but asking one to reconfigure ~/etc/dhcpd.conf. That is PI OS level and requires SUDO & amendment. I do not see how this help Klipper install

dw-0, 2024-01-28:
Im kind of siding with @jimmysdiy here. He made a good point. Im unsure if such features should be in the scope of KIAUHs feature set. It's not a bad idea, though not fitting KIAUHs desired scope i think. So for now i will label this issue as not planned, but keep it open for future tracking/referencing.

Nekrofago, 2024-02-13:
After your well explained comments and updating to bookworm I just realized that ~/etc/dhcpd.conf is no longer used to manage network connections by default.

I think this could be closed because even being simpler with NetworkManager CLI to set it up, is still out of scope.

closed 2024-02-17 by dw-0