dw-0/kiauh #425: RUN_SHELL_COMMAND not working with blank characters in parameter

Duffy77, 2024-01-25:
### Linux Distribution

OS: Raspbian GNU/Linux 11 (bullseye)
Distro: MainsailOS 1.3.2 (bullseye)

### What happened

I installed kiauh and enabled the shell command feature.
Wrote my macro to pass a text variable to a shell script.
With a simple word it works, but when i use blanks, i got the following error message
[image]

### What did you expect to happen

I would like to send use parameters with whitespaces

### How to reproduce

My shell_command.cfg
```
[gcode_shell_command GitBackup]
command: sh /home/pi/printer_data/config/test.sh
timeout: 30.
verbose: True
[gcode_macro GitBackup]
gcode:
    {% set commitmsg = params.COMMITMSG %}
    { action_respond_info("%s" % (commitmsg)) }
    RUN_SHELL_COMMAND CMD=GitBackup PARAMS={commitmsg}
```

my script test.sh
```
#!/bin/bash
if [ -n "$1" ]; then
    varname="$1"
fi

# Ask the user for their name
echo Hello, who am I talking to?

echo It\'s nice to meet you $varname

```

When i place a parameter with whitespaces like this 

[image]

i got the following error message
[image]

dw-0, 2024-01-27:
did you try and add quotation marks around `my test`?

Duffy77, 2024-01-27:
Hi,
When I put quotation marks around my test just like 'my test' then no error occurs direkt, but my ist passed as first parameter and I think test is the second .
But this is bad, because I would like a commit message to pass as first parameter

dw-0, 2024-01-27:
Another try would be to replace `$1` with `$@` in your bash script.

Duffy77, 2024-01-27:
This works :-) Thank you !

closed 2024-01-28 by Duffy77

Dalinaffeti, 2024-03-05:
Hi i am basically working around the same thing. I have a question after for example you have a g_code macro named: SET_PPR the param of ist named ppr. how can i use it in a g-code file. I mean how can i called and gived a value. Like example prr= 40.0