dw-0/kiauh #426: kiauh install problem on DietPI

Slawaspr0, 2024-01-27:
### Linux Distribution

DietPI + RaspberryPi 4

### What happened

Can't install kiauh. Can't run "cd ~ && git clone https://github.com/dw-0/kiauh.git" as normal user because no access, I relogin to root then run script, then when I'm using ./kiauh/kiauh.sh - getting error, can't run this as root, so I relogin back to "dietpi", then I can't run script because "dietpi" dont have access to "root/kiauh", its sick... I'm fighting with this few hours already

### What did you expect to happen

I needed to make new/clean system install on my RaspberryPi. Fast configuration - WIFI connection - SSH and trying to install kiauh. Last time i did it more than year ago. Something  changed since that, because I remember that last time I did it without problems.

### How to reproduce

Run DietPI
Login to user dietpi
Trying to run "cd ~ && git clone https://github.com/dw-0/kiauh.git" - cant, no access
Login to root
Again trying to run "cd ~ && git clone https://github.com/dw-0/kiauh.git" - success
Trying to run "./kiauh/kiauh.sh" - can't run this as root
Login to dietpi
Again trying to run "./kiauh/kiauh.sh"  - can't run, no directory access

### Additional information

What I was trying to do:

changing root/kiauh owner to dietpi - no success
then changing root/kiauh chmod to 0775 - still no success
script run but with a lot of errors

moving kiauh to home/dietpi/ location - no success
changing owner and chmod to 0775 to all files - no success
script run but with a lot of errors

What can I do? Can somebody help me?

dw-0, 2024-01-27:
I didn't test on dietpi for a long time, so no idea if and what has changed. dietpi is known to do some things differently.
but this is no issue with kiauh. it's an issue with the OS if you can't access a folder because of permission issues.
i suggest you just use a normal raspberry pi lite image instead.

Slawaspr0, 2024-01-27:
Thx for answer. I'm gonna try to write to DietPi, and maybe give them link to this topic. I'm curious what they will reply, if at all.

dw-0, 2024-01-27:
[image]

I just downloaded the DietPi for VMware and it works fine on my end.
You're aware that you need to create a new user with it's home directory and all that stuff with DietPi?

dw-0, 2024-01-27:
Ok sorry i was missing the point that you reported that the user dietpi is not working. It is on my end working too:
[image]

dw-0, 2024-01-27:
No issue in installing the full stack on a DietPi VM:
[image]

Slawaspr0, 2024-01-27:
Can't install this as dietpi user
[image]

Slawaspr0, 2024-01-27:
Already installed, failed because home/dietpi have root owner, not dietpi, thx for help!

dw-0, 2024-01-27:
Closed because solved in: https://github.com/MichaIng/DietPi/issues/6890

closed 2024-01-27 by dw-0