dw-0/kiauh #427: KlipperScreen did not start after install

shodanx2, 2024-01-27:
### Linux Distribution

debian 

### What happened

Installed klipperscreen via kiaut, install appears successful but it did not start

[image]

Now I can't seem to find the control to try and restart it

### What did you expect to happen

klipperscreen starts

### How to reproduce

Install klipperscreen via kiauth

shodanx2, 2024-01-27:
Ran

systemctl list-unit-files --type service -all | grep -i scr

To find the name of the process

[image]

Then ran 

systemctl  start KlipperScreen

to start the process

[image]

Still does not appear to work

[image]

I restart the computer, nochange

Ran kiauth.sh again, this time did the update (it wasn't very old but I don't update unnecessarily because updates break things)

Removed klipperscreen

[image]

re-Installed klipperscreen

[image]

[image]

[image]

Still the error

```
[✓ OK] KlipperScreen successfully installed!
find: ‘’: No such file or directory

###### Restart KlipperScreen.service ...
```

ChaosBlades, 2024-02-20:
Can confirm. Have to do this every time when installing KlipperScreen on Ubuntu Server.

`sudo bash -c "echo needs_root_rights=yes>>/etc/X11/Xwrapper.config"`

That needs to be added to the install script.

dw-0, 2024-02-20:
It looks like this issue has to be fixed by KlipperScreen. In case of KlipperScreen KIAUH is just executing KlipperScreens own install script: https://github.com/dw-0/kiauh/blob/master/scripts/klipperscreen.sh#L60

So please head over to the KlipperScreen repository and open an issue there, if there does not exist one already.

closed 2024-03-02 by dw-0