dw-0/kiauh #429: Hang Installing Crowsnest

john-clark, 2024-01-28:
### Linux Distribution

Raspberry Pi OS (legacy, 64-bit) Lite

Debian GNU/Linux 11 (bullseye) aarch64

Raspberry Pi 3 Model A Plus Rev 1.0

### What happened

Stopped on the following step for over an hour:

```bash
/=======================================================\
|     ~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~     |
|        Klipper Installation And Update Helper         |
|     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~     |
\=======================================================/

###### Cloning 'crowsnest' repository ...
[✓ OK] crowsnest repository already exists ...

###### Checking for the following dependencies:
● git
● make
[✓ OK] Dependencies already met!
Installer will prompt you for sudo password!

###### Launching crowsnest installer ...
crowsnest - A webcam daemon for multiple cams and stream services.

        Ahoy!
        Thank you for installing crowsnest ;)
        This will take a while ...
        Please reboot after the installation has finished.

Running apt-get update first ...

Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://deb.debian.org/debian bullseye-updates InRelease
Hit:3 http://security.debian.org/debian-security bullseye-security InRelease
Hit:4 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists...
Running apt-get update first ... [OK]
Doing some tests ...

Checking for mjpg-streamer ...

Checking for mjpg-streamer ... [OK]
Checking for camera-streamer dependencies ...

Checking if device is a Raspberry Pi ...

Checking if device is a Raspberry Pi ... [OK]
Checking if device is not a Raspberry Pi 5 ...

Checking if device is not a Raspberry Pi 5 ... [OK]
Checking if device is not running Ubuntu ...

Checking if device is not running Ubuntu ... [OK]
Checking for required kernel module ...

Checking for required kernel module ... [OK]
Checking for required packages ...

Checking for required packages ... [OK]
Checking for camera-streamer dependencies ... [OK]
Doing some tests ... [OK]
Installing with camera-streamer ...

[... 258 lines cut ...]
cc -std=gnu17 -MMD -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -c -o util/opts/opts.o util/opts/opts.c
g++ -std=c++17 -MMD -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -c -o device/libcamera/buffer.o device/libcamera/buffer.cc
g++ -std=c++17 -MMD -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -c -o device/libcamera/buffer_list.o device/libcamera/buffer_list.cc
g++ -std=c++17 -MMD -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -c -o device/libcamera/device.o device/libcamera/device.cc
g++ -std=c++17 -MMD -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -c -o device/libcamera/libcamera.o device/libcamera/libcamera.cc
g++ -std=c++17 -MMD -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -c -o output/rtsp/rtsp.o output/rtsp/rtsp.cc
g++ -std=c++17 -MMD -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -c -o output/webrtc/webrtc.o output/webrtc/webrtc.cc
```

### What did you expect to happen

Finish installing Crowsnest.

### How to reproduce

1. Install OS on supported sdcard using official installer
2. git clone kiauh and run install
3. choose install crowsnest
4. have running camera in mainsail

### Additional information

Normal running operation htop shows 102M in use and 39.8M swap and less than 2% use on each of the cores.

During the compile of camera-streamer on webrtc.cc the pi is unusable, ssh and web do not respond and Mem says 297M/419M and Swap 99.8M/100M cpu use on the 4 cores is less than 10% Load average 9.45 6.27 7.33

john-clark, 2024-01-28:
*Update*

Fresh install. *ONLY INSTALL* Crowsnest and it went through:

```bash
/=======================================================\
|     ~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~     |
|        Klipper Installation And Update Helper         |
|     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~     |
\=======================================================/

###### Cloning 'crowsnest' repository ...
Cloning into '/home/pi/crowsnest'...
remote: Enumerating objects: 1625, done.
remote: Counting objects: 100% (244/244), done.
remote: Compressing objects: 100% (110/110), done.
remote: Total 1625 (delta 155), reused 200 (delta 129), pack-reused 1381
Receiving objects: 100% (1625/1625), 632.25 KiB | 1.93 MiB/s, done.
Resolving deltas: 100% (982/982), done.

###### Checking for the following dependencies:
● git
● make
[✓ OK] Dependencies already met!
Installer will prompt you for sudo password!

###### Launching crowsnest installer ...
crowsnest - A webcam daemon for multiple cams and stream services.

        Ahoy!
        Thank you for installing crowsnest ;)
        This will take a while ...
        Please reboot after the installation has finished.

Running apt-get update first ...

Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Reading package lists...
Running apt-get update first ... [OK]
Doing some tests ...

Checking for mjpg-streamer ...

Checking for mjpg-streamer ... [OK]
Checking for camera-streamer dependencies ...

Checking if device is a Raspberry Pi ...

Checking if device is a Raspberry Pi ... [OK]
Checking if device is not a Raspberry Pi 5 ...

Checking if device is not a Raspberry Pi 5 ... [OK]
Checking if device is not running Ubuntu ...

Checking if device is not running Ubuntu ... [OK]
Checking for required kernel module ...

Checking for required kernel module ... [OK]
Checking for required packages ...

Checking for required packages ... [OK]
Checking for camera-streamer dependencies ... [OK]
Doing some tests ... [OK]
Installing with camera-streamer ...

Installing dependencies ...

Reading package lists... Done
Building dependency tree... Done
[... 1397 lines cut ...]
[ 97%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/websocketserver.cpp.o
[ 97%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/wstransport.cpp.o
[ 97%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/wshandshake.cpp.o
[100%] Linking CXX static library libdatachannel-static.a
make[5]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
[100%] Built target datachannel-static
make[4]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
make[3]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
make[2]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
g++ -Werror -Wall -g -I/home/pi/crowsnest/bin/camera-streamer -D_GNU_SOURCE -Wno-error=deprecated-declarations -DUSE_HW_H264 -DUSE_FFMPEG -DUSE_LIBCAMERA -I/usr/include/libcamera -DUSE_RTSP -I/usr/include/liveMedia -I/usr/include/groupsock -I/usr/include/BasicUsageEnvironment -I/usr/include/UsageEnvironment -DUSE_LIBDATACHANNEL -Ithird_party/libdatachannel/include -Ithird_party/libdatachannel/deps/json/include -o camera-streamer device/buffer.o device/buffer_list.o device/buffer_lock.o device/buffer_queue.o device/device.o device/device_list.o device/links.o output/http_ffmpeg.o output/http_h264.o output/http_hls.o output/http_jpeg.o output/output.o device/camera/camera.o device/camera/camera_debug.o device/camera/camera_decoder.o device/camera/camera_input.o device/camera/camera_isp.o device/camera/camera_output.o device/camera/camera_pipeline.o device/camera/camera_rescaller.o device/dummy/buffer.o device/dummy/buffer_list.o device/dummy/device.o device/dummy/dummy.o device/libcamera/fake_camera.o device/v4l2/buffer.o device/v4l2/buffer_list.o device/v4l2/debug.o device/v4l2/device.o device/v4l2/device_list.o device/v4l2/device_media.o device/v4l2/device_options.o device/v4l2/v4l2.o util/ffmpeg/remuxer.o util/http/http.o util/http/http_methods.o util/opts/control.o util/opts/fourcc.o util/opts/log.o util/opts/opts.o device/libcamera/buffer.o device/libcamera/buffer_list.o device/libcamera/device.o device/libcamera/libcamera.o output/rtsp/rtsp.o output/webrtc/webrtc.o html/index.html.o html/webrtc.html.o cmd/camera-streamer/http.o cmd/camera-streamer/main.o cmd/camera-streamer/opts.o cmd/camera-streamer/status.o third_party/libdatachannel/build/libdatachannel-static.a -lpthread -lstdc++ -lavcodec -lavformat -lavutil -lcamera -lcamera-base -lliveMedia -lgroupsock -lBasicUsageEnvironment -lUsageEnvironment -Lthird_party/libdatachannel/build -ldatachannel-static -Lthird_party/libdatachannel/build/deps/usrsctp/usrsctplib -lusrsctp -Lthird_party/libdatachannel/build/deps/libsrtp -lsrtp2 -Lthird_party/libdatachannel/build/deps/libjuice -ljuice-static -lcrypto -lssl
make[1]: Leaving directory '/home/pi/crowsnest/bin/camera-streamer'
Build 'camera-streamer' ... [SUCCESS]
Trying to set minimum GPU Memory to 128 MB ... [OK]

Do you want to add 'update manager' entry to your moonraker.conf? [Y/n]: Y
Trying to add 'update manager' entry to moonraker.conf ...
File 'moonraker.conf' does not exist! ... [SKIPPED]

Installation successful.

        Reboot your machine for the changes to take effect!

Reboot NOW? [y/N]: Y
Going to reboot in 5 seconds!
```

dw-0, 2024-01-28:
Well, i guess there is nothing to do. Seems like your pi simply hung up and now at the second try it worked.
I will close this issue then.

closed 2024-01-28 by dw-0

john-clark, 2024-01-29:
Yes, please close the issue as it does not seem to be an issue with kiauh.

I did try 5 times to compile camera-streamer.  I think it may be a gcc issue with the lower spec pi, as I put it in a 3b+ and the first 64 bit install, and compiled immediately. 

I am going to try and format and reinstall the 64bit os retry the install, to verify the original report, because if there is a bug I am sure you would like to document it.

eccentricexit, 2024-04-10:
> Yes, please close the issue as it does not seem to be an issue with kiauh.
> 
> I did try 5 times to compile camera-streamer. I think it may be a gcc issue with the lower spec pi, as I put it in a 3b+ with the first 64 bit install, and compiled immediately.
> 
> I am going to try and format and reinstall the 64bit os retry the install, to verify the original report, because if there is a bug I am sure you would like to document it.

Did you ever manage to install it?

It seems compiling is too much for my Pi 3b+ 64bit 1G RAM. Additional info accidentally installed the desktop instead of lite Raspberry Pi OS, so that is also taking up some resources.