dw-0/kiauh #430: RUN_SHELL_COMMAND params not being passed?

BETLOG, 2024-01-29:
### Linux Distribution

(mkspi)
Welcome to Armbian 22.05.0-trunk  with bleeding edge Linux 5.16.20-rockchip64

### What happened

```
$ FOCUSCAM0 RANGE=111
// Running Command {focus}...:
// $@:400
// $1:400
// Command {focus} finished
```

### What did you expect to happen

```
$ FOCUSCAM0 RANGE=111
// Running Command {focus}...:
// $@:111
// $1:111
// Command {focus} finished
```

### How to reproduce

----printer.cfg----
```
[gcode_shell_command focus]
command: sh /home/mks/set-v4l2-defaults.sh
timeout: 2
verbose: True

[gcode_macro FOCUSCAM0]
description:v4l2-ctl --dev /dev/v4l/by-id/usb-HD_WEB_CAMERA_HD_WEB_CAMERA_SN0001-video-index0 --set-ctrl=focus_absolute=<range/1024>
gcode:
    #SET_GCODE_VARIABLE MACRO=FOCUSCAM0 VARIABLE=range VALUE=222
#    {% set range = 400 %}
#    {% set range = params.range|default(400)|int %}
   #{% set range = variable.range|default(200)|int %}
    #{% set range = ''|default(200) %}
#    {% set range = params.EXTRUDER_TEMP|default(222, true)|int %}
#    {% set range = params.range|default(400, true)|int %}
    {% set range = params.range|default(400)|int %}
    #{ action_respond_info("%s" % (range)) }
    RUN_SHELL_COMMAND CMD=focus PARAMS={range}
```

----/home/mks/set-v4l2-defaults.sh----
```
#!/bin/sh
#is sh NOT bash
[ $1 -gt 0 ] && range=$1 || range=400
echo "\$@:$@"
echo "\$1:$1"
v4l2-ctl --dev /dev/v4l/by-id/usb-HD_WEB_CAMERA_HD_WEB_CAMERA_SN0001-video-index0 --set-ctrl=focus_automatic_continuous=0
v4l2-ctl --dev /dev/v4l/by-id/usb-HD_WEB_CAMERA_HD_WEB_CAMERA_SN0001-video-index0 --set-ctrl=focus_absolute=$range
```

### Additional information

It's entirely possible this is caused by my own stupidity, as this jinja syntax is new to me.

dw-0, 2024-01-29:
```diff
[gcode_macro FOCUSCAM0]
description:v4l2-ctl --dev /dev/v4l/by-id/usb-HD_WEB_CAMERA_HD_WEB_CAMERA_SN0001-video-index0 --set-ctrl=focus_absolute=<range/1024>
gcode:
    #SET_GCODE_VARIABLE MACRO=FOCUSCAM0 VARIABLE=range VALUE=222
--- {% set range = params.range|default(400)|int %}
+++ {% set range = params.RANGE|default(400)|int %}
    RUN_SHELL_COMMAND CMD=focus PARAMS={range}
```

Although this is rather a support request for klipper, i can help you with that case.
You need to use `params.RANGE` inside of the macro. An uppercase "range" is required here. Then, it should work.

Funfact: When calling the macro it doesn't matter if you call it like `FOCUSCAM0 RANGE=111` or `FOCUSCAM0 range=111`. When calling a macro it seems that parameters can be written in lower- or uppercase, but inside of macros, you need to strictly use uppercase.

BETLOG, 2024-01-29:
> +++ {% set range = params.RANGE|default(400)|int %}

Excellent, that was it. Thanks.

> Although this is rather a support request for klipper

I only ask here becasue [gcode_shell_command.py](https://github.com/dw-0/kiauh/blob/master/resources/gcode_shell_command.py) is a subset of this git.

closed 2024-01-29 by BETLOG