dw-0/kiauh #436: Mobileraker permission issue causes notification warning

shodanx2, 2024-02-06:
### Linux Distribution

debian 12

### What happened

Getting this error message in MainSail and on KlipperScreen

[image]

I have 6 instances installed

[image]

Installed with basically default settings these

Klipper
Moonraker
Mainsail
Fluidd
KlipperScreen
Crowsnest
OctoEverywhere
Mobileraker
Octoprint

### What did you expect to happen

Not have the warning message

### How to reproduce

Regular installation of moonraker

### Additional information

Fix is simply to add the word

mobileraker

to each instance's file

/home/user/printer_1_data/moonraker.asvc

After a reboot, the message is gone

Tomblarom, 2024-03-03:
My `moonraker.asvc` looked like this. I highly suspect a new-line is missing somewhere :D
[image]

dw-0, 2024-03-23:
I think that is something that should be handled by Mobileraker?
@Clon1998 are you aware of this or did you already resolved that issue?

Clon1998, 2024-03-23:
Hey,
yes that is on me. I was finally able to reproduce that issue and fix it.

Related to https://github.com/Clon1998/mobileraker_companion/issues/84.

dw-0, 2024-03-23:
Great! Thank you for the fast response!

closed 2024-03-23 by dw-0

shodanx2, 2024-03-23:
Who decided to use invisible coding characters in a settings file !? 🤦

dw-0, 2024-03-23:
> Who decided to use invisible coding characters in a settings file !? 🤦

Not sure what or who you are blaming exactly now, but using newline characters is quite common in text manipulation scenarios.

shodanx2, 2024-03-23:
It's a tongue in cheek dig at yaml, don't worry. 

How EOF should be considered equivalent to EOL.