dw-0/kiauh #437: I am trying to use the shell_command feature from Kiauh

Dalinaffeti, 2024-02-07:
### Linux Distribution

Raspbian

### What happened

i wanted to execute the typical testing "echo hello world with the following:
[gcode_shell_command hello_world]
command: sudo echo "hello world"
timeout: 2.
verbose: True
[gcode_macro HELLO_WORLDD]
gcode:
    RUN_SHELL_COMMAND CMD=hello_world

but when i type: HELLO_WORLDD in the console i ddint get an output back

### What did you expect to happen

i expected the output hello world

### How to reproduce

thats something that i dont know. I dont think this is a bug, maybe i am just missing on something

dw-0, 2024-02-08:
There is no point in using sudo for a simple echo command. Try without sudo.

Dalinaffeti, 2024-02-08:
i tried without but it is the same thing. Even when i give the HELP command on the console nothing shows up

Dalinaffeti, 2024-02-08:
[image]

dw-0, 2024-02-08:
Not sure what the issue could be if even the HELP command does not output anything.
Please consider joining the Klipper or Mainsail Discord, to get support for that. I will close this issue, as this is not really related to an issue with KIAUH.

closed 2024-02-08 by dw-0