dw-0/kiauh #442: Mainsail nginx fails to install if IPV6 is disabled -- Raspbian Bookworm

RickoT, 2024-02-14:
### Linux Distribution

Rasbian Bookworm x64 Lite Rasberry Pi Image

### What happened

During the installation of Mainsail this error is received
```
Created symlink /etc/systemd/system/multi-user.target.wants/nginx.service → /lib/systemd/system/nginx.service.
Could not execute systemctl:  at /usr/bin/deb-systemd-invoke line 145.
Setting up nginx (1.22.1-9) ...
Job for nginx.service failed because the control process exited with error code.
See "systemctl status nginx.service" and "journalctl -xeu nginx.service" for details.
invoke-rc.d: initscript nginx, action "start" failed.
× nginx.service - A high performance web server and a reverse proxy server
     Loaded: loaded (/lib/systemd/system/nginx.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Wed 2024-02-14 08:10:45 PST; 19ms ago
       Docs: man:nginx(8)
    Process: 2979 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=1/FAILURE)
        CPU: 10ms

Feb 14 08:10:45 ender-purple systemd[1]: Starting nginx.service - A high performance web server and a reverse proxy server...
Feb 14 08:10:45 ender-purple nginx[2979]: nginx: [emerg] socket() [::]:80 failed (97: Address family not supported by protocol)
Feb 14 08:10:45 ender-purple nginx[2979]: nginx: configuration file /etc/nginx/nginx.conf test failed
Feb 14 08:10:45 ender-purple systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Feb 14 08:10:45 ender-purple systemd[1]: nginx.service: Failed with result 'exit-code'.
Feb 14 08:10:45 ender-purple systemd[1]: Failed to start nginx.service - A high performance web server and a reverse proxy server.
dpkg: error processing package nginx (--configure):
 installed nginx package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.11.2-2) ...
Errors were encountered while processing:
 nginx
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

### What did you expect to happen

Installation completes without issue

### How to reproduce

Fresh imaging of Raspi OS (bookworm) Lite x64
Edit /boot/firmware/cmdline.txt and add ipv6.disable=1

Go through installation process with kiuah.sh and install mainsail

RickoT, 2024-02-14:
in order to successfully complete installation I need to remove ipv6.disable=1 from cmdline.txt.  I dont use ipv6 and don't want it if I dont need it.

dw-0, 2024-02-14:
You can disable ipv6 after the installation again.

The issue is the following:
While installing the nginx package for the first time on a system via `apt-get install nginx`, it automatically installs a default configuration in `/etc/nginx/sites-available/default` and links the config via a symlink to `/etc/nginx/sites-enabled/default`. During the installation the nginx.service gets started. The issue is, that the default config contains this:
```
server {
        listen 80 default_server;
        listen [::]:80 default_server;
```
See the second "listen" config? That's what is causing the issue. The starting nginx.service loads the default config, ipv6 is disabled although the config expects it to be enabled, and fails. Which in turn results in an exit code of 1 of the package installation, thus killing KIAUH.

I currently see no way of preventing the NGINX service from starting during its install, so i can handle the start of the service myself, after the mainsail config has been installed AND the default nginx config is disabled/deleted.

RickoT, 2024-02-14:
Yup, and that's what I've done, i just wanted to pass it along

closed 2024-02-17 by dw-0