dw-0/kiauh #449: Install pyserial with python

alidaf, 2024-03-12:
### Is your feature request related to a problem? Please describe

I'm trying to install firmware to an ERCF, which needs pyserial. Unfortunately, I cannot install it because I run into an error: externally-managed-environment

### Describe the solution you'd like

To be able to use pyserial

### Describe alternatives you've considered

Installing globally - fails

dw-0, 2024-03-23:
Hi, im unsure what exactly KIAUHs role in this issue is or should be. Can you elaborate?

alidaf, 2024-03-24:
I am thinking that KIAUH is a universal installer for all things Klipper and the de-facto standard for installation and management, so making sure all dependencies are catered for would be a consistent approach. I think some firmware flashing can already be carried out by KIAUH if I'm not mistaken. I know this is a bit of an outlier but since KIAUH already installs Python it would be simpler and more convenient given that Python seems to now expect venv based installations, which are outside of my comfort zone. This was more of an issue when trying to install pyserial on Armbian than it was an RaspiOS but package managers like pip are a little more nuanced than apt and I've come unstuck with it many times, especially when dealing with different versions of Python.

dw-0, 2024-03-31:
Hi! Thanks for explaining, but im sorry i still don't understand what the actual feature request for KIAUH would be at this point. I don't have an ERCF, hence im unable to relate to anything that is required to set this up or flash any involved microcontrollers. I had a brief look into the ERCF manual https://github.com/EtteGit/EnragedRabbitProject/blob/main/Documentation/ERCF_Manual.pdf to try and get an idea of what your feature request could be. Is it the requirement for a custom flash command which differs from the ones currently supported in KIAUH? But that still wouldn't answer the question in regards to pyserial and why you aren't able to install it.

closed 2024-04-06 by dw-0