dw-0/kiauh #460: camera-streamer on rpi5

m-anish, 2024-04-27:
### Is your feature request related to a problem? Please describe

Hi. Firstly, wonderful project. Thank you! 

I was wondering if rpi5 can run camera-streamer, because while installing crowsnest on rpi5, the installation skips installing camera-streamer

### Describe the solution you'd like

camera-streamer support on rpi5 if possible

mryel00, 2024-04-27:
First of all this issue has nothing to do with this Repo.
> I was wondering if rpi5 can run camera-streamer

That question would suit the camera-streamer repo. In theory, yes it can.
> because while installing crowsnest on rpi5, the installation skips installing camera-streamer

That's right, because there is no reason for camera-streamer to exist in context of Crowsnest on a Pi5.
To explain this:
The Pi5 has no HW encoders, besides the H265. camera-streamer uses HW encoders for everything regarding Pi cameras and for WebRTC. So the only use of camera-streamer would be USB cams with a MJPG stream.
But why should Crowsnest install two streamers doing exactly the same? As ustreamer is doing this job more stable than camera-streamer, there is currently no reason to install camera-streamer.

You most likely wrote this message, because of the missing Pi camera support on the Pi5 with Crowsnest. If you would look at the Crowsnest repo issues, you would find this: https://github.com/mainsail-crew/crowsnest/issues/241
That's a FR for spyglass support and the currently "official" workaround on how to use Pi cameras on a Pi5.

If you asked this because you want WebRTC, there is basically no way of getting a WebRTC stream, that's supported by all browsers, on a Pi5, without running at full CPU usage.

Please ask such questions in the repo, of the correct project. KIAUH is not deciding what to install, it's the user deciding it. There are no hidden flags, that will prevent e.g. camera-streamer from installing with Crowsnest. Those are things the repo, here Crowsnest, decides on it's own.

m-anish, 2024-04-28:
Opened https://github.com/mainsail-crew/crowsnest/issues/262

closed 2024-04-28 by dw-0