dw-0/kiauh #461: Make possible to set dir to install

SlimRG, 2024-04-30:
### Is your feature request related to a problem? Please describe

I don't want to have 100500+ reps in home dir, I want to have only one dir 'printer' with apps 

### Describe the solution you'd like

Add path selection in settings

dw-0, 2024-04-30:
Thanks for the request. I added it to the backlog. It will not be implemented in the current version 5 of KIAUH. I consider to implement it in version 6, as the migration from bash to python will drastically help in that regard.

dw-0, 2024-05-01:
I just saw that this is actually a duplicate of [#372](https://github.com/dw-0/kiauh/issues/372), so i will close this issue for now as it is already been tracked in my backlog :)

closed 2024-05-01 by dw-0

dw-0, 2024-05-03:
Sorry, the backlog is not public.