dw-0/kiauh #463: Copy instances/endpoints info while updating mainsail/fluidd

szafran81, 2024-05-02:
### Is your feature request related to a problem? Please describe

Right now every update removes instances/endpoints configuration from config.json in both mainsail and fluidd.

### Describe the solution you'd like

While updating read and copy to new config the instances configuration:

In mainsail (config.json):
```
    "instancesDB": "json",
    "instances": [
        { "hostname": "192.168.1.99", "port": 7125 },
        { "hostname": "192.168.1.99", "port": 7126 }
    ]
```

In fluidd (config.json):
```
  "endpoints": [
    "http://192.168.1.99:7125",
    "http://192.168.1.99:7126"
  ],
```

And I think there is a lot more info that can be copied with that.
Maybe walking through the config and only copying from new config.json data that's been added and leaving old settings untouched?

### Describe alternatives you've considered

Reconfiguring by manually editing config files after every update is such a waste of time.

dw-0, 2024-05-02:
Thanks for the request. I added it to the backlog. It will not be implemented in the current v5 of KIAUH. I consider to implement it in v6, as the migration from bash to python will drastically help in that regard.

szafran81, 2024-05-03:
That's great. Thank you for taking it into consideration.

dw-0, 2024-08-31:
This is fixed with KIAUH v6 which's alpha got released just a minute ago.

closed 2024-08-31 by dw-0