dw-0/kiauh #464: Cloning Klipper from  https://github.com/Klipper3d/klipper :  failed!

fabse-hack, 2024-05-04:
### Linux Distribution

MainSail OS

### What happened

kiauh Klipper uninstall / install - install failed

### What did you expect to happen

-

### How to reproduce

install klipper ...

### Additional information

i try to install klipper with kiauh:

why i get this error?

```

###### Checking for the following dependencies:
● git
[✓ OK] Dependencies already met!

###### Cloning Klipper from https://github.com/Klipper3d/klipper ...
Cloning into '/home/pi/klipper'...
remote: Enumerating objects: 60994, done.
remote: Counting objects: 100% (47/47), done.
remote: Compressing objects: 100% (34/34), done.
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
error: 4450 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: index-pack failed

#=======================================================#
 Cloning Klipper from
 https://github.com/Klipper3d/klipper
 failed!
#=======================================================#

pi@mainsailos:~ $
```

dw-0, 2024-05-04:
So... what should **I** do, if even you yourself think it is a git problem?

closed 2024-05-04 by dw-0

xavimc, 2025-01-15:
To all of you having issues downloading klipper from git.
I tried 1000 different ways and git configs but I was always trying with raspbian / raspbian lite 64bits.
I was about to give up and went for raspbian lite 32bits and everything went smooth...

Also after installing ...first thing I did:
sudo apt update
sudo apt full-upgrade
sudo apt install git

Good luck!