dw-0/kiauh #466: Set name to access in local network

julianCast, 2024-05-07:
### Is your feature request related to a problem? Please describe

When we reboot our router, the IP changes, I know we could fix the IP to avoid this, but a name would be convenient.

### Describe the solution you'd like

For example mainsailOS expose `mainsailos.local ` in the first boot, [here](https://docs-os.mainsail.xyz/getting-started/first-boot#access-mainsail).

### Describe alternatives you've considered

Setting a fixed IP.

### Additional information

I understand I'm comparing the issue with a OS image, but I just wonder if it could be something doable.
Thank you for the fantastic work you guys do.

dw-0, 2024-05-07:
It's already possible.
From the main menu, go to the advanced menu, then select option 7 `[Change hostname]`. This will install the `avahi-daemon` and changes the hostname of the host.

julianCast, 2024-05-07:
> It's already possible. From the main menu, go to the advanced menu, then select option 7 `[Change hostname]`. This will install the `avahi-daemon` and changes the hostname of the host.

Oh I missed it! Thank you so much and sorry for the issue. Ill close the door