dw-0/kiauh #483: Update menu error when Spoolman not installed

BeansIsFat, 2024-06-25:
### Linux Distribution

Debian GNU/Linux 11 (bullseye)

### What happened

Update Menu has the following error when Spoolman is not installed:
```jq: error: Could not open file /home/pi/spoolman/release_info.json: No such file or directory```

### What did you expect to happen

Update Menu should display without errors regardless of whether Spoolman is installed.

### How to reproduce

* Install `jq` prerequisite to avoid other errors.
* Use KIAUH v5.0.0-85 on any machine that has never had Spoolman installed.
* Navigate to Update Menu.

BeansIsFat, 2024-06-26:
Fixed in 7993b98

closed 2024-06-26 by BeansIsFat

ACampy, 2024-06-26:
After pulling in your latest update I'm still having what I think is this same issue.
I do not have spoolman installed and having JQ installed or not doesn't seem to matter. Attached a screenshot of what my update menu looks like.
[image]