dw-0/kiauh #489: Missing nacl Libs

kevinroosa, 2024-07-31:
### Linux Distribution

Raspbian OS Lite (64-bit), Build Date 4 July 2024.

### What happened

"Moonraker failed to connect" error on attempting to load the web UI.

`journalctl -u moonraker.service --pager-end`

returned "OSError: Could not locate nacl lib, searched for libsodium."

### What did you expect to happen

Web UI load after using KIAUH for Klipper/Moonraker installation.

### How to reproduce

1. Installed Pi OS Lite (64-bit)
2. Used commands provided in readme to complete installation.
3. Rebooted, waited 5 minutes, then attempted to connect (tried both IP and local address, no dice).

### Additional information

Current workaround seems to be:

`sudo apt-get install python3-nacl`

After reboot, web UI loads as expected. Possibly just a quick apt edit?

DNAScanner, 2024-08-01:
I just encountered the same issue, but installing nacl manually using `sudo apt-get install python3-nacl` fixed the issue for me. Credits: https://stackoverflow.com/questions/72710756/could-not-locate-nacl-lib-searched-for-libsodium

wildjerry, 2024-08-02:
Seconding this, I ran into the same issue. 

`sudo apt install libsodium23` also works.

wildjerry, 2024-08-02:
If I'm understanding correctly, KIAUH pulls moonraker deps from the official [moonraker install script](https://github.com/Arksine/moonraker/blob/master/scripts/install-moonraker.sh). The relevant part of that script should be

``` bash
        PACKAGES="python3-virtualenv python3-dev libopenjp2-7 libsodium-dev zlib1g-dev"
        PACKAGES="${PACKAGES} libjpeg-dev packagekit wireless-tools curl"
        PACKAGES="${PACKAGES} build-essential"
```

I don't have a pi in front of me, can someone check if maybe the `libsodium-dev` package doesn't exist on the official apt repository? If so, I think the preferred solution would be to change the install script on the moonraker side, but should be possible to just add an extra dependency from the KIAUH script.

Would also appreciate it if someone more experienced with KIAUH's scripts could confirm that I've understood this correctly.

kevinroosa, 2024-08-03:
> If I'm understanding correctly, KIAUH pulls moonraker deps from the official [moonraker install script](https://github.com/Arksine/moonraker/blob/master/scripts/install-moonraker.sh). The relevant part of that script should be
> 
> ```shell
>         PACKAGES="python3-virtualenv python3-dev libopenjp2-7 libsodium-dev zlib1g-dev"
>         PACKAGES="${PACKAGES} libjpeg-dev packagekit wireless-tools curl"
>         PACKAGES="${PACKAGES} build-essential"
> ```
> 
> ~I don't have a pi in front of me, can someone check if maybe the `libsodium-dev` package doesn't exist on the official apt repository? If so, I think the preferred solution would be to change the install script on the moonraker side, but should be possible to just add an extra dependency from the KIAUH script.~
> 
> `libsodium-dev` exists, and depends on libsodium23, meaning my theory was wrong.
> 
> Would also appreciate it if someone more experienced with KIAUH's scripts could confirm that I've understood how KIAUH gets dependencies correctly.

My line of thinking currently is that there may be missing Python bindings? The only dep for libsodium is libc6. I'll be replicating the events on a fresh build in the next day or two, hopefully it shines a little light on the issue.

IceBoxWillys, 2024-08-03:
Just wanted to add that I ran into the same problem. wildjerry's fix worked for me and after a reboot the web is up and running as expected.

Unfortunately my understanding of the mechanics of this failure are pretty limited, but wanted to speak up in case this is a broader issue with a new version of kiuah

Thanks!

dw-0, 2024-08-03:
> Would also appreciate it if someone more experienced with KIAUH's scripts could confirm that I've understood how KIAUH gets dependencies correctly.

@wildjerry you are correct. KIAUH pulls the dependencies from the install script.

I could need some more info, `Raspbian OS Lite (64-bit), Build Date 4 July 2024` is a good info, but do you guys can tell which Debian Version those images base on? There are Bookworm and Bullseye images available, both in 64-bit and both with build dates of 4th of july. Would be interesting to know if this is something only happening on Bookworm or Bullseye.

dw-0, 2024-08-03:
So, while trying to reproduce #491 i installed Moonraker as well. The installation went through totally fine but when i checked the status of the service, it was exited. I checked the moonraker.log and i see the same error log you guys reported. This is on a Pi 5 with Raspberry Pi OS Lite 64-bit Debian 12 (Bookworm).

```
-------------------- Log Start | Sat Aug  3 07:00:51 2024 --------------------
platform: Linux-6.6.31+rpt-rpi-2712-aarch64-with-glibc2.36
data_path: /home/pi/printer_data
is_default_data_path: False
config_file: /home/pi/printer_data/config/moonraker.conf
startup_warnings: []
verbose: False
debug: False
asyncio_debug: False
is_backup_config: False
is_python_package: False
instance_uuid: 7015670030ff4fdc9feae5dfe7d06a5e
unix_socket_path: /home/pi/printer_data/comms/moonraker.sock
software_version: v0.9.2-0-gb40751b
git_branch: master
git_remote: origin
git_repo_url: https://github.com/Arksine/moonraker.git
modified_files: []
unofficial_components: []
log_file: /home/pi/printer_data/logs/moonraker.log
python_version: 3.11.2 (main, May  2 2024, 11:59:08) [GCC 12.2.0]
launch_args: /home/pi/moonraker-env/bin/python /home/pi/moonraker/moonraker/moonraker.py -d /home/pi/printer_data
msgspec_enabled: False
uvloop_enabled: False
2024-08-03 09:00:51,777 [confighelper.py:read_file()] - Configuration File '/home/pi/printer_data/config/moonraker.conf' parsed, total size: 446 B
2024-08-03 09:00:51,777 [server.py:add_log_rollover_item()] -
#################### Moonraker Configuration ####################

[server]
host = 0.0.0.0
port = 7125
klippy_uds_address = /home/pi/printer_data/comms/klippy.sock

[authorization]
trusted_clients =
        10.0.0.0/8
        127.0.0.0/8
        169.254.0.0/16
        172.16.0.0/12
        192.168.0.0/16
        FE80::/10
        ::1/128
cors_domains =
        *.lan
        *.local
        *://localhost
        *://localhost:*
        *://my.mainsail.xyz
        *://app.fluidd.xyz

[octoprint_compat]

[history]

[update_manager]
channel = dev
refresh_interval = 168

#################################################################
All Configuration Files:
/home/pi/printer_data/config/moonraker.conf
#################################################################
2024-08-03 09:00:51,799 [server.py:load_component()] - Component (secrets) loaded
2024-08-03 09:00:51,801 [server.py:load_component()] - Component (template) loaded
2024-08-03 09:00:51,802 [server.py:load_component()] - Component (klippy_connection) loaded
2024-08-03 09:00:51,968 [server.py:load_component()] - Component (application) loaded
2024-08-03 09:00:51,974 [server.py:load_component()] - Component (websockets) loaded
[... 31 lines cut ...]
    server.load_components()
  File "/home/pi/moonraker/moonraker/server.py", line 269, in load_components
    self.load_component(config, component)
  File "/home/pi/moonraker/moonraker/server.py", line 298, in load_component
    module = importlib.import_module(full_name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/pi/moonraker/moonraker/components/authorization.py", line 21, in <module>
    from libnacl.sign import Signer, Verifier
  File "/home/pi/moonraker-env/lib/python3.11/site-packages/libnacl/__init__.py", line 87, in <module>
    nacl = _get_nacl()
           ^^^^^^^^^^^
  File "/home/pi/moonraker-env/lib/python3.11/site-packages/libnacl/__init__.py", line 83, in _get_nacl
    raise OSError(msg)
OSError: Could not locate nacl lib, searched for libsodium.so, libsodium.so.23, libsodium.so.18, libsodium.so.17, libsodium.so.13, libsodium.so.10, libsodium.so.5, libsodium.so.4,
2024-08-03 09:00:52,008 [server.py:main()] - Server Shutdown
```

closed 2024-08-03 by dw-0

wildjerry, 2024-08-03:
Just in case it still matters, I ran into the issue on Pi 4, not 5.

dw-0, 2024-08-03:
Yes, after a quick chat with the moonraker main developer, we noticed that the moonraker install script has changed, which broke kiauhs capabilit of parsing it correctly. So the issue actually is not hardware or image specific at all, like it first had seemed. I simply did not notice the change in the moonraker install script.

I pushed a fix to KIAUH, can everyone please try again and maybe report back if it works again? It should, i tested it on a fresh system and it worked for me.

kevinroosa, 2024-08-03:
Just ran a fresh test from the same OS Lite Minimal image (64-Bit Bookworm) and all is well. No errors reported in moonraker.service.

Thanks for the quick turnaround on this one!

PR #492 (2024-08-03): fix: parse moonraker dependencies from system-dependencies.json
A recent moonraker update broke KIAUHs parsing capabilities for the system dependencies.
This PR will fix the parsing and uses a different file for parsing the dependencies.
As soon as more distros are added to the file, this method needs to be checked again.

Fixes #489