dw-0/kiauh #491: Virtualenv error on fresh bookworm install

PhilippMolitor, 2024-08-02:
### Linux Distribution

Raspberry Pi 5, Latest Raspbian Lite Bookworm 64bit

### What happened

When installing klipper, this error occurrs:

```
###### Installing Python 3.11.2 virtual environment...
created virtual environment CPython3.11.2.final.0-64 in 202ms
  creator CPython3Posix(dest=/home/pi/klippy-env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/pi/.local/share/virtualenv)
    added seed packages: pip==23.0.1, setuptools==66.1.1, wheel==0.38.4
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
/home/pi/kiauh/scripts/klipper.sh: line 283: /home/pi/klippy-env/bin/pip: No such file or directory
```

### What did you expect to happen

The `pip` command should be available. I tried changing the kiauh klipper.sh script to use `python -m venv ...` and use activate/deactivate, with that it works, but that of course changes the behavior.

### How to reproduce

Install fresh Raspbian Lite Bookworm 64bit on a Raspberry Pi 5, then KIAUH and try to install klipper.

dw-0, 2024-08-03:
Hi, so i tried to reproduce the issue but sadly i wasn't able to.
[image]
[image]

The Klipper installation wen't through all the way as you can see on the second screenshot, and the service is up and running.

PhilippMolitor, 2024-08-03:
I reinstalled the base OS, then it worked... maybe this is some weird edge case.

dw-0, 2024-08-03:
Thanks for reporting back. That is indeed really weird. I will close this issue, if a similar issue arises in the future, please open a new issue.

closed 2024-08-03 by dw-0