dw-0/kiauh #497: allow offline install from local github repositories for local installation and network resource management

MarioOrtegon, 2024-08-15:
### Is your feature request related to a problem? Please describe

It's related to github's routing issue of August 14th 2024, I kindly request to have another advanced option for local installation in a user configurable path for pointing to the right github DB. current error details below:
[image]

### Describe the solution you'd like

allow user to type local repository in advanced section to install dependencies locally offline

### Describe alternatives you've considered

I've tried installing klipper from local repository but don't know how to properly create it's enviroment so kiauh can still use it once github is online or device is back on network access.

### Additional information

error log:

/=======================================================\
|     ~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~     |
|        Klipper Installation And Update Helper         |
|     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~     |
\=======================================================/

###### Initializing Klipper installation ...

/=======================================================\
| Please select your preferred Python version.          |
| The recommended version is Python 3.x.                |
|-------------------------------------------------------|
|  1) [Python 3.x]  (recommended)                       |
|  2) [Python 2.7]  (legacy)                            |
|-------------------------------------------------------|
|                       B) « Back                       |
\=======================================================/
###### Select Python version: 1
   [➔] Python 3.x

/=======================================================\
| Please select the number of Klipper instances to set  |
| up. The number of Klipper instances will determine    |
| the amount of printers you can run from this host.    |
|                                                       |
| WARNING:                                              |
| Setting up too many instances may crash your system.  |
|-------------------------------------------------------|
|                       B) « Back                       |
\=======================================================/
###### Number of Klipper instances to set up: 1
   [➔] Instance count: 1

###### Installing single Klipper instance ...

###### Checking for the following dependencies:
● git
[✓ OK] Dependencies already met!

###### Cloning Klipper from https://github.com/Klipper3d/klipper ...
Cloning into '/home/mayos/klipper'...
remote: Enumerating objects: 64931, done.
remote: Counting objects: 100% (58/58), done.
remote: Compressing objects: 100% (26/26), done.
error: RPC failed; curl 56 Recv failure: Connection reset by peer
error: 3957 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output

#=======================================================#
 Cloning Klipper from
 https://github.com/Klipper3d/klipper
 failed!
#=======================================================#

mayos@SidewinderX1:~ $pip install fuck (o)bligatory online dependencies

dw-0, 2024-08-15:
Hi,

thanks for the request. FWIW i don't see it as KIAUHs scope for mitigating Github issues of any kind.
There are no such things as local github repositories, but i think i get what you mean and you simply mean local git repositories, in other terms: git repos that have been cloned previously and still exist on your filesystem?

> hopefully we can also remove printer/enviroment individually

This is actually something i have developed for the upcoming KIAUH v6 release. You will be able to select which components to remove during removal. Here is a sneak peek on how the Klipper remove menu will look like. Other remove menus will look similar:
[image]

I think technically it could be possible to re-install e.g. Klipper from scratch even if there is no internet connection. Provided that all required resources are cached (Python packages) and local git repositories still exist at their expected location with their expected names. I will have a look at it, but this will not be anything released for KIAUHs current v5.

MarioOrtegon, 2024-08-15:
Hi thank you for the quick reply and yes I do have all dependencies updated
on RPi and cloned klipper repo on it from another pc so some terminal
screenshot examples could be useful in the wiki if someone wants to try
manually on v5/6. Thank you I wasn't aware of that new implementation but
look forward using it, reduces dependency on terminal/ssh. Thank you for
looking into it, hopefully it's not too cumbersome to adapt since that way
we could modify cfg files before installation so kiauh knows which custom
configuration to use from its own interface, I know it's easy to dream and
ask for new features but it's a very reliable essential 3d tool so hope you
don't exceed your self too much and have a nice evening

Hans-KK, 2024-09-21:
One way to handle this issue could be to implement an option to let kiauh download from github via ssh instead of https.
Like: git@github.com:Klipper3d/klipper.git instead of https://github.com/Klipper3d/klipper.git
To make it work one will need a github account and setup some keys to allow ssh download from github.

I got same issue and I can not download klipper via https, but via ssh it works fine.

dw-0, 2024-09-21:
> download from github via ssh instead of https.

I think i don't understand your comment.
"download" and "offline" are kind of mutually exclusive.
apart from that, downloading via ssh should work by now (with kiauh v6).
you just have to change the download url of klipper via the settings menu.

Hans-KK, 2024-09-21:
Sorry that I got it sort of wrong, my situation was like I was searching for an offline solution to install klipper, and I found this thread, and at the same time I was looking into ssh download from github, so I was not attentive, sorry.

kerry-c-young, 2024-11-02:
I've had this issue too. For some reason I can't pull the repo successfully. What I ended up doing was downloading the zipped repo, unzipping into /home/user/klipper and running the installation again. When it gives you an option to overwrite the directory, just say no and it will skip the clone and continue on.
Hope this helps

wang098138, 2024-11-03:
Sorry, I am new to installing KIAUH V6 and I keep encountering this message.
Obviously it's not normal. Can you help? Thanks.

[ERROR] Error installing packages:
E: Unable to correct problems, you have held broken packages.

[ERROR] Error during installation of Klipper requirements!
[ERROR] Command '['sudo', 'apt-get', 'install', '-y', 'python3-virtualenv', 'avr-libc', 'gcc-avr', 'dfu-util', 'avrdude', 'stm32flash', 'libnewlib-arm-none-eabi', 'binutils-arm-none-eabi', 'gcc-arm-none-eabi', 'libncurses-dev', 'libusb-1.0', 'libusb-dev', 'binutils-avr', 'virtualenv', 'libffi-dev', 'python3-dev']' returned non-zero exit status 100.
[ERROR] Klipper installation failed!

sierrapappabravo, 2024-11-03:
> I've had this issue too. For some reason I can't pull the repo successfully. What I ended up doing was downloading the zipped repo, unzipping into /home/user/klipper and running the installation [...]

Was having the same problem. Your solution worked! Thanks!

xavimc, 2025-01-15:
To all of you having issues downloading from klipper from git.
I tried 1000 different ways and git configs but I was always trying with raspbian / raspbian lite 64bits.
I was about to give up and went for raspbian lite 32bits and everything went smooth...

Also after installing ...first thing I did: 
sudo apt update
sudo apt full-upgrade
sudo apt install git

Good luck!