dw-0/kiauh #509: Using new KIUAH, fresh armbian system, when fluidd-config install fails

GoldenGreek, 2024-09-01:
### Linux Distribution

armbian-mkspi debian bullseye (6.6.17-current-rockchip64) running on mks-skipr-mini board. Repo is [here](https://github.com/redrathnure/armbian-mkspi)

### What happened

1.  Fresh base-image from the repo above with device-tree tweaked for the mini version of the board.
2. Installed kiauh.sh and ran the beta/python version.
3. Installed klipper & moonraker without issue
4. While installing fluidd and the default fluidd config, i receive the following error:
```
###### Downloading Fluidd from https://github.com/fluidd-core/fluidd/releases/latest/download/fluidd.zip ...
Downloading: [####################]100.00% (3.59/3.59MB)
[OK] Download complete!

###### Extracting fluidd.zip ...
[OK] OK!

###### Add section '[update_manager fluidd]' to '/home/mks/printer_data/config/moonraker.conf' ...

###### Restart moonraker.service ...
[OK] OK!

###### Downloading Fluidd-Config ...

###### Cloning repository from 'https://github.com/fluidd-core/fluidd-config.git'
Cloning into '/home/mks/fluidd-config'...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (28/28), done.
remote: Total 46 (delta 24), reused 38 (delta 16), pack-reused 0 (from 0)
Receiving objects: 100% (46/46), 40.23 KiB | 2.12 MiB/s, done.
Resolving deltas: 100% (24/24), done.
[OK] Clone successful!

###### Create symlink for fluidd.cfg ...

###### Linking /home/mks/fluidd-config/fluidd.cfg to /home/mks/printer_data/config

###### Creating backup of config-printer_data in /home/mks/kiauh-backups/printer-cfg-backups ...
[OK] Backup successful!

###### Add section '[update_manager fluidd-config]' to '/home/mks/printer_data/config/moonraker.conf' ...
[ERROR] Fluidd-Config installation failed!
[Errno 18] Invalid cross-device link: '/tmp/tmp7co34fzc' -> '/home/mks/printer_data/config/printer.cfg'

###### Creating NGINX config for Fluidd ...
[OK] NGINX config for Fluidd successfully created.

###### Link NGINX logs into log directory ...

###### Restart nginx ...
[OK] OK!

[OK] Fluidd installation complete!
Open Fluidd now on: http://192.168.126.192:81
```
5. NGINX is running, and site is accessible.

> Other info:  /tmp is mounted as tmpfs

### What did you expect to happen

I expected the fluidd config to install properly.

### How to reproduce

Fresh install of armbian-mkspi and all components.  I suspect this is an issue with /tmp being a tmpfs on this platform.

GoldenGreek, 2024-09-01:
FYI, works fine with the current release (non-beta) version.

dw-0, 2024-09-01:
> I suspect this is an issue with /tmp being a tmpfs on this platform.

I suspect the same, but not sure how to actually tackle this issue right now.
Can you try disabling tmpfs for testing purposes and try reproduce the issue? If the issue then does not occur, we can say for sure it is caused by tmpfs.

GoldenGreek, 2024-09-02:
Sorry for the late reply.  Unfortunately, I'm very busy at the moment and will attempt this tomorrow after work, and will let you know the results.

GoldenGreek, 2024-09-04:
My first attempt failed, and when I checked /tmp was no longer mounted as tmpfs, but as zram2, so I had to disable that as well.  For thoroughness here are the steps I did to test this per @dw-0's request:

1) Removed /tmp as tmpfs from fstab.
2) Ran `sudo systemctl mask tmp.mount` to ensure that /tmp doesn't mount as tmpfs.
3) Edited /etc/default/armbian-zram-config and set `ENABLED=false`
4) Rebooted and verified /tmp is not tmpfs
5) Cleanly uninstalled all items that were installed using KIUAH
6) Using KIAUH v6, reinstalled all items, including fluidd/fluidd config without issue.

Our suspicions have been confirmed.

Thanks!

GoldenGreek, 2024-09-10:
@dw-0  Do you need any more from me on this report?

dw-0, 2024-09-11:
> @dw-0 Do you need any more from me on this report?

Hi thanks for providing some more information. I hope this will help. I have to somehow try and reproduce, then fix the issue.

dw-0, 2024-09-22:
Hi, i just released KIAUH v6.0.0-alpha.4.
For that release i reworked the config handling. Config files are now opened, read and written a bit different. Maybe this works with tmpfs filesystems now. Can you please try again and see if it works now?

loonatec, 2024-09-24:
I just tried to install Fluidd on a fresh Armbian install using KIAUH v6.0.0-alpha.4 and it still encounters the same issue.
This is a Fly Gemini v3 board running the Armbian stable image from [reemo3dp](https://github.com/reemo3dp/mellowfly-geminipi-armbian), following the instructions from [DerFuchs](https://github.com/DerFuchs/Armbian-with-Klipper-on-Fly-Gemini-3).

The first time through I get the same error:
`[Errno 18] Invalid cross-device link: '/tmp/tmp7co34fzc' -> '/home/mks/printer_data/config/printer.cfg'`

I tried disabling tmpfs but probably misunderstood the fix as it put the system into a read-only state. 

After starting over and trying the KIAUH v6.0.0-alpha.4 version and encountering the same issue I then tried uninstalling Fluidd and reinstalling both Fluidd and Mainsail but encounter the following error (same for both):
```
╔═══════════════════════════════════════════════════════╗
║ ~~~~~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~~~~~ ║
║        Klipper Installation And Update Helper         ║
║ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ║
╚═══════════════════════════════════════════════════════╝
╔═══════════════════════════════════════════════════════╗
║ ~~~~~~~~~~~~~~~~~~~ [ Main Menu ] ~~~~~~~~~~~~~~~~~~~ ║
╟──────────────────┬────────────────────────────────────╢
║  0) [Log-Upload] │   Klipper: Installed: 1            ║
║                  │     Owner: Klipper3d               ║
║  1) [Install]    │      Repo: klipper                 ║
║  2) [Update]     ├────────────────────────────────────╢
║  3) [Remove]     │ Moonraker: Installed: 1            ║
║  4) [Advanced]   │     Owner: Arksine                 ║
║  5) [Backup]     │      Repo: moonraker               ║
║                  ├────────────────────────────────────╢
║  S) [Settings]   │        Mainsail: Not installed     ║
║                  │          Fluidd: Not installed     ║
║ Community:       │   Client-Config: -                 ║
║  E) [Extensions] │                                    ║
║                  │   KlipperScreen: Not installed     ║
║                  │       Crowsnest: Not installed     ║
╟──────────────────┼────────────────────────────────────╢
║  v6.0.0-alpha.4  │  Changelog: https://git.io/JnmlX   ║
╟──────────────────┴────────────────────────────────────╢
║                        Q) Quit                        ║
╚═══════════════════════════════════════════════════════╝
###### Perform action: 1
╔═══════════════════════════════════════════════════════╗
║ ~~~~~~~~~~~~~~~ [ Installation Menu ] ~~~~~~~~~~~~~~~ ║
╟───────────────────────────┬───────────────────────────╢
║ Firmware & API:           │ Touchscreen GUI:          ║
║  1) [Klipper]             │  7) [KlipperScreen]       ║
║  2) [Moonraker]           │                           ║
║                           │ Webcam Streamer:          ║
║ Webinterface:             │  8) [Crowsnest]           ║
║  3) [Mainsail]            │                           ║
║  4) [Fluidd]              │                           ║
║                           │                           ║
║ Client-Config:            │                           ║
║  5) [Mainsail-Config]     │                           ║
║  6) [Fluidd-Config]       │                           ║
╟───────────────────────────┴───────────────────────────╢
║                       B) « Back                       ║
╚═══════════════════════════════════════════════════════╝
###### Perform action: 4

┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃                       [ INFO ]                        ┃
┠───────────────────────────────────────────────────────┨
┃ It is recommended to use special macros in order to   ┃
┃ have Fluidd fully functional and working.             ┃
┃                                                       ┃
┃ The recommended macros for Fluidd can be seen here:   ┃
┃ https://github.com/fluidd-core/fluidd-config          ┃
┃                                                       ┃
┃ If you already use these macros skip this step.       ┃
┃ Otherwise you should consider to answer with 'Y' to   ┃
┃ download the recommended macros.                      ┃
┗━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┛
###### Download the recommended Fluidd-Config? (Y/n): 

###### Downloading Fluidd from https://github.com/fluidd-core/fluidd/releases/latest/download/fluidd.zip ...
Downloading: [####################]100.00% (3.62/3.61MB)
[OK] Download complete!

###### Extracting fluidd.zip ...
[OK] OK!

###### Add section '[update_manager fluidd]' to '/home/robert/printer_data/config/moonraker.conf' ...
[ERROR] Fluidd installation failed!
'str' object has no attribute 'keys'

```

dw-0, 2024-09-26:
The `'str' object has no attribute 'keys'` issue should be fixed now with the new alpha 5 release.
I still was not able to setup a tmpfs filesystem myself and further investigate the original issue.

4eIIIuP, 2024-10-15:
Good evening. Trying to install klipper via kiauh v6.0.0.0-alpha.7, but encountered the same problem when installing fluidd
```
[OK] OK!
[ERROR] Fluidd-Config installation failed!
[Errno 18] Invalid cross-device link: '/tmp/tmpoxl84ce6' -> '/home/mks/printer_data/config/printer.cfg'
```
MKS SKIPR board, ermbian taken from here https://github.com/redrathnure/armbian-mkspi/releases?ysclid=m27xw28jrw918440727.

Translated with DeepL.com (free version)

dw-0, 2024-10-21:
I was able to reproduce the issue and found a solution. It will be released with alpha.9

PR #584 (2024-10-24): fix: Path.rename() not working across devices
causes `[Errno 18] Invalid cross-device link` on tmpfs filesystems

fixes #509
PR #590 (2024-10-24): Merge develop into master (Release v6.0.0-alpha.9)
fixes #509 
fixes #514
fixes #576 
fixes #585
closed 2024-10-24 by dw-0