dw-0/kiauh #512: Cannot change Klipper Repository

eriksonssilva, 2024-09-02:
### Linux Distribution

Debian 12

### What happened

When selecting "1) Set Klipper source repository" and inserting the new repository, nothing happens - It still uses the klipper3d repository:

[image]

### What did you expect to happen

Changing the repository for the custom one.

### How to reproduce

Follow the UI:
Main Screen > S for Settings > 1 for changing Klipper repo > Fill in the information

### Additional information

I have even tried editing scripts/globals.sh and default.kiauh.cfg, updating them with the repo I wanted, but it still somehow manages to grab the repo from klipper3d.

Thanks a lot in advance!

DizzieNight, 2024-09-02:
+1 been having the same issue

eriksonssilva, 2024-09-02:
@DizzieNight 
What I did meanwhile was: uninstall everything and delete all the folders (if I did not delete the folders, the setup would not work as intended).
Then
`git checkout 8547942` (this is where I assumed things were still working).

Then proceeded with the normal installation process and it is working properly.

dw-0, 2024-09-02:
> I have even tried editing scripts/globals.sh and default.kiauh.cfg

You decided to use KIAUH v6. `scripts/globals.sh` is not used by v6.
You edited the `default.kiauh.cfg`. The changelog explicitly states to NOT modify this file. 
The changelog also mentions how to do it correctly.

> Then `git checkout 8547942` (this is where I assumed things were still working).

You could have simply choose to start KIAUH v5 in the startup dialog instead of checking out this specific commit.
This is not a solution and other users should not follow this "advise", as it leaves you in a detached head state, and you won't receive updates in this state.

eriksonssilva, 2024-09-02:
Hey @dw-0 ! I didn't mean to disappoint or to cause any issue to others.
I have just done what felt more straightforward to me, and I must confess I did not read the changelog. I just wanted something working ASAP, since I wanted/needed to print something and it was already late in the night here :)

I have done what the changelog says (deleting the line version_to_launch= from .kiauh.ini), and I was able chosing version 5.
Everything works as intended now!

Once again, I meant no harm with my "advice". It's just that many people like me will first come read the issues and only after (if at all) they read the changelog. Lesson learned, my mistake once again.

Thanks!

dw-0, 2024-09-02:
Hi, no problem. I just wanted to point you to the right direction.

I reopened this issue as your bugreport was actually valid, there IS a bug with the v6 repo switch feature, it does not work correctly. So let's leave this issue open so it is tracked here :)

PR #519 (2024-09-05): fix: fix switching of repositories
Extend the functionality of repo switching by creating a backup before the switch. Also implement a rollback mechanic in case of an error.

Fixes #512
closed 2024-09-08 by dw-0