dw-0/kiauh #513: Cant start kiauh after install to v6. Updated all git and OS withous any change. Checked python version 3.8.10

islonina, 2024-09-03:
### Linux Distribution

Linux Mint 20

### What happened

After updating i got errors cousing not loading to main menu

### What did you expect to happen

Loading main menu of kiauh

### How to reproduce

Updating to v6 from old kiauh

### Additional information

Very old multi instance install but updated from time to time

dw-0, 2024-09-03:
Thanks for the report, i was able to reproduce the issue.
It happens as soon as the moonraker.conf contains something like this:
```
[update_manager mainsail]
#type: web
#type: web
channel: stable
repo: mainsail-crew/mainsail
path: /home/dw/mainsail
```
This needs to be fixed.

islonina, 2024-09-04:
Yes. Deliting this part on all instances i was able to run kiauh and reinstlaing moonraker  without a problem. Thx a lot

flavio-costa, 2024-09-13:
I had a similar problem, after updating, at startup I chose option 3 and I only have this screen when opening kiauh.
[image]

closed 2024-09-22 by dw-0

eduncan911, 2024-10-10:
@dw-0 i'm facing a similar situation. my moonraker does not have a crowsnest updating block.  could you point me in another direction?

```
╔═══════════════════════════════════════════════════════╗
║ ~~~~~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~~~~~ ║
║        Klipper Installation And Update Helper         ║
║ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ║
╚═══════════════════════════════════════════════════════╝
[ERROR] An unexpected error occured:
cannot unpack non-iterable NoneType object
Traceback (most recent call last):
  File "/home/klipper/kiauh/kiauh/core/menus/base_menu.py", line 184, in run
    self.display_menu()
  File "/home/klipper/kiauh/kiauh/core/menus/base_menu.py", line 178, in display_menu
    self.print_menu()
  File "/home/klipper/kiauh/kiauh/core/menus/main_menu.py", line 127, in print_menu
    self._fetch_status()
  File "/home/klipper/kiauh/kiauh/core/menus/main_menu.py", line 96, in _fetch_status
    self._get_component_status("cn", get_crowsnest_status)
  File "/home/klipper/kiauh/kiauh/core/menus/main_menu.py", line 100, in _get_component_status
    status_data: ComponentStatus = status_fn(*args)
  File "/home/klipper/kiauh/kiauh/components/crowsnest/crowsnest.py", line 157, in get_crowsnest_status
    return get_install_status(CROWSNEST_DIR, files=files)
  File "/home/klipper/kiauh/kiauh/utils/common.py", line 127, in get_install_status
    org, repo = get_repo_name(repo_dir)
TypeError: cannot unpack non-iterable NoneType object
```

eduncan911, 2024-10-11:
Per the latest changelog that included v6.0.0-alpha, I was able to revert by deleting the `version_to_launch=` line in ~/.kiauh.ini