dw-0/kiauh #515: Possible misspelling of "origin" and "managed_services" in moonraker conf.

simpat1zq, 2024-09-03:
### Linux Distribution

BTT CB2 board with their latest image. Debian based I believe. 

### What happened

I am using the alpha version that just came out. I installed everything from kiauh. I just finished getting my printer.cfg setup, and moonraker gave me warnings. The first one was:

```
 Moonraker warnings found.
    [update_manager]: Failed to load extension KlipperScreen: No option 'origin' in section: 'update_manager KlipperScreen'
```
 I opened up the moonraker config and found that 'origin was spelled 'orgin'. I fixed it and got a new error:
 
```
  Moonraker warnings found.
    Unparsed config option 'manages_servcies: KlipperScreen' detected in section [update_manager KlipperScreen]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
```
So that 'manages_servcies' should be 'managed_services' so it's got a couple of typos in there. After I fixed that, there were no more errors. This only happened in KlipperScreen. I installed fluidd and crowsnest as well, and they were all good. 

### What did you expect to happen

config settings to be spelled correctly. 

### How to reproduce

Just install KlipperScreen via kiauh. 

### Additional information

I typed this up quickly. Let me know if you need any more info from me, but I think this should be one of the easiest bugs you'll fix today(unless it's something funky that happened only on my machine).

thhdragon, 2024-09-04:
mainsail-config in the moonraker config has a few typos also. manages instead of managed and orgin instead of origin

dw-0, 2024-09-07:
> mainsail-config in the moonraker config has a few typos also. manages instead of managed and orgin instead of origin

Hey, i wasn't able to find the typo you are mentioning for the mainsail-config. I did some tests, removed all configs and made clean installs so all configs and config entries get recreated. The moonraker.conf looked like this after installing the mainsail-config:
[image]
This looks correct to me.

The other typos are indeed an issue and will be fixed soon.

Samfire54, 2024-09-08:
Got the exact same issue, and got exactly the same fix. Good to go now

PR #520 (2024-09-08): fix: typo in "origin" and "managed_services"
Fixes #515
closed 2024-09-08 by dw-0