dw-0/kiauh #516: No webinterface after using kaiuh

alcomys, 2024-09-04:
### Linux Distribution

Debian 11 Bullseye

### What happened

After installing klipper/moonraker/octopi no webinterface.
This is the 3rd try to use that tool kiauh but i think you cannot use it. 

### What did you expect to happen

I expect that i have a webinterface after the installs

### How to reproduce

Installing klipper/moonraker/octopi

### Additional information

Almost no information on how to use kiauh not even the menu documentation.
very less information on how to solve problems.

dw-0, 2024-09-04:
> but i think you cannot use it.

I think hundreds, if not thousands of other people think different.

How do you think the information you provided can help me figure out what your actual issue is?

alcomys, 2024-09-04:
My problem is repeated with the same actions.
So chose in the KIAUH menu to install first klipper than moonraker than octoprint or Mainsail all the system files are up to date, but no web interface, is did try experimental version of KIAUH also , but the same issue with no web interface. 
So if the method described in the less documents are not working what can i do about it?
I did exact what has been written and it is still not working, I became crazy from it. 
And I'm not even an noob at raspberry pi and other Linux distro's, So what do we need than? an engineer to install klipper etc?

I'm sorry to say, your tool is a really nice looking tool, but working with it is a struggle.

dw-0, 2024-09-04:
Your information still does not help me in any way. It does not help if you just say: "It does not work".
I  have not seen what was going on in your terminal during any of the installation steps, nor do i see any of the current state your system is in. Giving any support right now is impossible.

> So if the method described in the less documents are not working what can i do about it?
I did exact what has been written and it is still not working, I became crazy from it.

I don't know which documentation you are referring to.

> I'm sorry to say, your tool is a really nice looking tool, but working with it is a struggle.

Feel free to do all of the required steps for installing Klipper, Moonraker and a webinterface of your choice without KIAUH.
If you're experienced with Linux, you should be able to get all of those components working without any issues at all just by following their respective installation documentation. KIAUH is "just" automating all of those steps.

Check if the Klipper service is running.
Check if the Moonraker service is running.
Check if the NGINX service is running (assuming you installed Mainsail/Fluidd)
And maybe explain how you try to access the webinterface.

alcomys, 2024-09-04:
Hello, 

All the installers are running without error in the terminal so there is nothing to see an error or so.
Klipper and Moonraker is running.
I did not do this step : Check if the NGINX service is running (assuming you installed Mainsail/Fluidd) but that is the case that there is no error support or some of check list. and i don't see Octoprint in your suggestion. I  see nothing that you have to install NGINX . Maybe that is the step that is missing in the instructions?
Or some of an check to make shore that all is running?

Thanks 

My web interface is https://My-ip I also did some check in my router to find the right web interface but it got nothing to show.

dw-0, 2024-09-04:
Octoprint does not require NGINX.

> Maybe that is the step that is missing in the instructions?

I still don't understand which instructions you are referring to.

> My web interface is `https://my-ip/`

During all installations of any webinterface, you were shown URLs on how to access those.
Do not use https. Use http.
Mainsail/Fluidd will be served on `http://<ip>`
Octoprint will be served on `http://<ip>:5000`

alcomys, 2024-09-04:
Hello DW, 

I did try everything, but can't get it running. 
I was trying for over 5 day's reading and reading but no result. 
All the options you bring are not working at my setup. sorry

I want to thank you for your help, but i did switch back to Marlin. 
And i leave the klipper thing to others.

closed 2024-09-05 by dw-0