dw-0/kiauh #518: listing MCU's with USART when none exist results in incorrect menu path

c0psrul3, 2024-09-04:
### Linux Distribution

MainsailOS / pi-z2

### What happened

typed "b" to go back, to main menu, but was presented with menu for selection of flash command

### What did you expect to happen

expected to return to "Advanced Menu"

### How to reproduce

1.  select Advanced  `4`
2. select Get MCU ID `4`
3. select USART with no devices connected `2`
4. see error
```sh
/bin/sh: 1: Syntax error: word unexpected (expecting ")")
[ERROR] Unable to find a UART device!
Command '"find /dev -maxdepth 1 -regextype posix-extended -regex "^\/dev\/tty(AMA0|S0)$" 2>/dev/null"' returned non-zero exit status 2.
[WARN] No MCUs found!
[WARN] Make sure they are connected and repeat this step.
```
5. input menu states
```
###### Select MCU to flash: 
```
6. select Back `b`
7. experience wrong menu
```
╔═══════════════════════════════════════════════════════╗
║ Make sure that the controller board is connected now! ║
╟───────────────────────────────────────────────────────╢
║ How is the controller board connected to the host?    ║
╟───────────────────────────────────────────────────────╢
```

[image]

dw-0, 2024-09-04:
This is the expected behavior.
When no MCUs could be detected because lets say you forgot to connect it, you can simply go back one step, connect the MCU, and select the connection type again. The identification of MCUs will then start again.

closed 2024-09-07 by dw-0

3b, 2024-10-04:
I'm also seeing this problem. Going back 1 step works, but if you don't want to try again (or if you do and it doesn't work as in #557), it takes 2 extra steps to get back to `[ Advanced Menu ]` from `How is the controller board connected to the host?`

normal flow:
```
[ Advanced menu ]
4 -> How is the controller board connected to the host? 
B -> [ Advanced menu ]
```

after failing to find UART:
```
[ Advanced menu ]
4 -> How is the controller board connected to the host? 
2 -> [List of detected MCUs]
B -> How is the controller board connected to the host? 
B -> * Which flash command to use for flashing the MCU?
B -> * [ MCU Flash Menu ] 
B -> [ Advanced Menu ]
```
entries marked with `*` are not `back` from `How is the controller board connected to the host?`

When `1) USB` successfully finds something, it stays at `How is the controller board connected to the host?`. Not sure if that is intended or not, but selecting `B` from there goes back to `[ Advanced Menu ]` as expected.