dw-0/kiauh #528: Support for non-systemd environments

apparle, 2024-09-11:
### Is your feature request related to a problem? Please describe

Kiauh is a very convenient set of installation tools, but it is locked-in with systemd.
When trying to set up lightweight environments like Android Termux proot-distro environment, or alpine linux, or a lightweight alpine based docker container, systemd is just not available. 
I'm trying to use an old Android phone which has everything (USB-OTG support, a good camera,a nice touchscreen) that will be very good to run `Klipper+Moonraker+Mainsail+Klipperscreen`. With termux and termux-x11 there's all the tools needed to run any reasonable app. But it being an older kernel and a proot env, I only have openrc or sysv init services and not proper systemd; and trying to install systemd is near impossible because the kernel is derived off of Android or docker or other env.

I see the past issue from years ago https://github.com/dw-0/kiauh/issues/287 which isn't clear on the solution -- Is the stance a flat no?

### Describe the solution you'd like

There should be a mechanism to instruct KIAUH script to install everything as standalone apps without the services, and then generate standalone `start.sh` & `stop.sh` which the user can invoke or manually install set up in their own service mechanisms as needed?
Or alternatively, support for openrc or sysvinit or other non-systemd service infrastructures?

### Describe alternatives you've considered

The current alternative is to give up on Kiauh altogether and manually install Klipper, Moonraker, Mainsail one by one. Definitely not easy.

CODeRUS, 2024-09-14:
solution was not removing init;d scripts, you are free to write your own autorun scripts for your favorite init system.

fyi: https://github.com/mkuf/prind

dw-0, 2024-09-15:
Thank you for the request. I see that there is demand for support on non-systemd environments. However, supporting new base distros besides Debian is possible, but maintaining it may be a burden. Especially for such niche usecases like using old android phones as a host system. For docker, i can recommend prind as well, although i haven't tested it on my own, the project seems really nice and most important is actively maintained.

I plan to support more base systems in the future (Fedora, Arch, etc.) but i have no timeframe for it yet and this decision is also not final and may be subject to change. So for now, i tagged this issue as a feature request and that further evaluation is required.

apparle, 2024-09-16:
> solution was not removing init;d scripts, you are free to write your own autorun scripts for your favorite init system.
> 
> fyi: https://github.com/mkuf/prind

But how do I get Kiauh to not try to install systemd dependencies or interact with systemd?
Right now, I've an Ubuntu environment setup inside of Android's terminal with chroot (or proot, I tried both), and Kiauh tries to pull-in a lot of systemd packages which then fails because its a chroot.
I don't mind writing my own autorun scripts, in fact I precisely want to do that -- I just need a way to tell Kiauh explicitly with some conf option or env variable `DONT_USE_SYSTEMD_INIT` which indicates:
- `This system is not based on systemd`, don't try that at all.
- And may be an optional `Here's a directory with template systemd init scripts dumped, adapt to your own init system`. 

Is there already a way to do this? Am I doing something completely wrong?
I understand that implementing support for more base systems is a lot of work both for implementation & maintenance, so maybe above is a reasonable first step ?

Some more thoughts on proper support for different base systems (Just my naive mental model, can't say I understand all complexities yet):
- Getting the right dependencies installed, with some different commands (Eg: `apk add` instead of `apt install` for alpine) and different package naming convensions. If you carve out package name & commands to separate files (text, json, python etc.), others in community (like me) can post PRs to update/add/maintain the dependency commands/package names for the base system they care for. The repo can claim official support only for debian based systems with everything else as alpha/beta.
- Getting the right init system hooks to autorun / start / stop: Above option of just "disable systemd" is a good start. Only if we want to go further, it can be modularized into separate commands + paths, others in the community can help maintain their own fav init system they care about. The repo stance can still be only systemd is official, everything else is alpha/beta. 

The reason I'm suggesting this -- Kiauh is popular enough that underlying projects (Klipper, Moonraker, Mainsail) all suggest Kaiuh as the right way to install things, and it's hard to find standalone installation instructions anywhere; even for the savvy. I literally ran Kiauh on a standalone system + read code to reverse engineer the install for these and then set them up manually. Since it is popular enough already, maybe extend it to be friendly for all linux based systems :-)

assasukasse, 2025-03-22:
I am also in the same situation, albeit slightly different.
I am running antix, which uses sysV init or runit, kiauh fails looking for systemd.
I think the solution could be as easy as providing a choice: do you want to install systemD scripts? and if not, a folder is created with some template script that the user could adapt to its environment...

etoimene, 2025-10-22:
Plus one.