dw-0/kiauh #539: Repo name extending outside of the main menu "box"

T9Air, 2024-09-22:
### Linux Distribution

ubuntu server 22.04

### What happened

I am using a custom Klipper repo, and the name extends outside of the main menu box. I believe that this did not happen in earlier versions, and instead the name was cut off at the end.

### What did you expect to happen

The name should be cut off if it is too long.

### How to reproduce

Use a custom repository that has a long name.

### Additional information

[image]

dw-0, 2024-09-22:
Hi, thanks for the report.
Those names were never shortened or cut. But they will with the next release :)

closed 2024-09-22 by dw-0

T9Air, 2024-09-22:
Thanks! I just do not remember noticing it before.

PR #540 (2024-09-22): fix: trunc owner and repo name if they would overflow
fixes #539