dw-0/kiauh #545: KIAUH 6 - Upload klipper via DFU fails

cad435, 2024-09-24:
### Linux Distribution

 Armbian 24.5.1 Bookworm with Linux 6.6.32-current-meson64 on BananaPi CM4

### What happened

When I try to upload a new klipper.bin so a STM32 in DFU mode KIAUH quits with the following error:

```
[ERROR] Unable to find a USB DFU device!
Command '"lsusb | grep "DFU" | cut -d " " -f 6 2>/dev/null"' returned non-zero exit status 127.
[WARN] No MCUs found!
[WARN] Make sure they are connected and repeat this step.
```

### What did you expect to happen

An upload to the device

### How to reproduce

Place a STM32 Board with Hardware Bootloader into DFU mode. 
Try to upload klipper.bin via KIAUH

### Additional information

If I manually type the command
 `lsusb | grep "DFU" | cut -d " " -f 6 2>/dev/null` 
 
 Into the console I get
`0483:df11` 
which would be the correct value for uploading.

the exact same procedere is working flawlessly with KIAUH v5

tameike000, 2024-09-29:
Can confirm this:
Bullseye RPi3 with 6 alpha: no DFU flash possible -> not found. Same with Bookworm.
With 5 works like a charm.
If you need logs, let me know.

dw-0, 2024-10-03:
Hi @tameike000 and @cad435, sorry for the late response, i haven't had time to look into this issue yet.
Can you both please provide the output of `lsusb` please? I don't have any USB devices i can flash with DFU, but i think having the raw output of the unmodified/unpiped `lsusb` command might already help. Thanks in advance!

dw-0, 2024-10-03:
FYI: i may have already a patch ready. If you are familiar with manually checking out different git branches, you could try and test if my patch works. The name of the branch is `fix/KIA-545` and you can check it out with:
```sh
cd ~/kiauh
git fetch origin fix/KIA-545
git checkout fix/KIA-545
```

after testing make sure you check out master again, so you receive regular updates to KIAUH again
```sh
cd ~/kiauh
git checkout master
```

closed 2024-10-05 by dw-0

dw-0, 2024-10-05:
Hi, thanks for your feedback.
Yeah the actual issue in KIAUH was, that the DFU device was not identified correctly, because of a bug how that identification was handled.

The reason KIAUH says that flashing failed is, that the command did exit with an error code. Probably because detaching the device failed and then the whole command fails, although the flash already succeeded. I don't think i can do something about it as i can only check for the exit code and then distinguish if the action succeeded or might have failed.

commit dc026a7a (2024-10-05): Release v6.0.0-alpha.6

Merge develop into master (v6.0.0-alpha.6)

fixes #545
fixes #553
fixes #557