dw-0/kiauh #546: Weird python errors after CAN upgrade

szafran81, 2024-09-26:
### Linux Distribution

Distributor ID: Ubuntu
Description:    Ubuntu 24.04.1 LTS
Release:        24.04
Codename:       noble

### What happened

I've upgraded my Voron 0.1 to 0.2r1 with CAN

### What did you expect to happen

For it to work properly

### How to reproduce

Don't really know.

### Additional information

Hi,

I've upgraded my V0 to a CAN setup and after updating my mcu config with the proper canbus_uuid's I'm getting this error:
[image]

My setup is on a x86 mini PC with Lubuntu LTS installed, and Klipper/Moonraker/Mainsail installed via Kiauh v5.

After getting thet I've tried this:
```
szafran@3DP:~$ sudo apt update ; sudo apt upgrade ; sudo apt dist-upgrade ; sudo apt autoremove ; sudo apt autoclean
Stary:1 http://ftp.vectranet.pl/ubuntu noble InRelease
Stary:2 https://download.docker.com/linux/ubuntu noble InRelease
Stary:3 http://ftp.vectranet.pl/ubuntu noble-updates InRelease
Stary:4 http://ftp.vectranet.pl/ubuntu noble-security InRelease
Stary:5 http://ftp.vectranet.pl/ubuntu noble-backports InRelease
Stary:6 https://dl.google.com/linux/chrome/deb stable InRelease
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności... Gotowe
Odczyt informacji o stanie... Gotowe
5 pakietów może być zaktualizowanych. Można je zobaczyć wykonując 'apt list --upgradable'.
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności... Gotowe
Odczyt informacji o stanie... Gotowe
Obliczanie aktualizacji... Gotowe
Get more security updates through Ubuntu Pro with 'esm-apps' enabled:
  libcjson1 libavdevice60 ffmpeg libpostproc57 libavcodec60 libavutil58
  libswscale7 libswresample4 libavformat60 libavfilter9
Learn more about Ubuntu Pro at https://ubuntu.com/pro
Aktualizacje odroczone, z powodu stopniowego wprowadzania przez dystrybucję:
  python3-distupgrade python3-update-manager ubuntu-release-upgrader-core ubuntu-release-upgrader-qt update-manager-core
0 aktualizowanych, 0 nowo instalowanych, 0 usuwanych i 5 nieaktualizowanych.
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności... Gotowe
Odczyt informacji o stanie... Gotowe
Obliczanie aktualizacji... Gotowe
Get more security updates through Ubuntu Pro with 'esm-apps' enabled:
  libcjson1 libavdevice60 ffmpeg libpostproc57 libavcodec60 libavutil58
  libswscale7 libswresample4 libavformat60 libavfilter9
Learn more about Ubuntu Pro at https://ubuntu.com/pro
Aktualizacje odroczone, z powodu stopniowego wprowadzania przez dystrybucję:
  python3-distupgrade python3-update-manager ubuntu-release-upgrader-core ubuntu-release-upgrader-qt update-manager-core
0 aktualizowanych, 0 nowo instalowanych, 0 usuwanych i 5 nieaktualizowanych.
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności... Gotowe
Odczyt informacji o stanie... Gotowe
0 aktualizowanych, 0 nowo instalowanych, 0 usuwanych i 5 nieaktualizowanych.
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności... Gotowe
Odczyt informacji o stanie... Gotowe
```

Everything under kiauh is up to date also.

I've tried reinstalling Klipper/Moonraker via Kiauh - nothing changed.

I've tried this:
```
systemctl stop klipper-Mikrus
~/klippy-env/bin/pip install -r ~/klipper/scripts/klippy-requirements.txt
cd ~/klippy-env/bin
./pip install --upgrade python-can==4.3
systemctl start klipper-Mikrus
```
result:
```
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ====
Wymagane jest uwierzytelnienie, aby zatrzymać jednostkę „klipper-Mikrus.service”.
Authenticating as: Szafran (szafran)
Password:
==== AUTHENTICATION COMPLETE ====
Ignoring greenlet: markers 'python_version < "3.12"' don't match your environment
Requirement already satisfied: cffi==1.14.6 in ./klippy-env/lib/python3.12/site-packages (from -r /home/szafran/klipper/scripts/klippy-requirements.txt (line 5)) (1.14.6)
Requirement already satisfied: pyserial==3.4 in ./klippy-env/lib/python3.12/site-packages (from -r /home/szafran/klipper/scripts/klippy-requirements.txt (line 6)) (3.4)
Requirement already satisfied: greenlet==3.0.3 in ./klippy-env/lib/python3.12/site-packages (from -r /home/szafran/klipper/scripts/klippy-requirements.txt (line 8)) (3.0.3)
Requirement already satisfied: Jinja2==2.11.3 in ./klippy-env/lib/python3.12/site-packages (from -r /home/szafran/klipper/scripts/klippy-requirements.txt (line 9)) (2.11.3)
Requirement already satisfied: python-can==3.3.4 in ./klippy-env/lib/python3.12/site-packages (from -r /home/szafran/klipper/scripts/klippy-requirements.txt (line 10)) (3.3.4)
Requirement already satisfied: markupsafe==1.1.1 in ./klippy-env/lib/python3.12/site-packages (from -r /home/szafran/klipper/scripts/klippy-requirements.txt (line 11)) (1.1.1)
Requirement already satisfied: pycparser in ./klippy-env/lib/python3.12/site-packages (from cffi==1.14.6->-r /home/szafran/klipper/scripts/klippy-requirements.txt (line 5)) (2.22)
Requirement already satisfied: wrapt~=1.10 in ./klippy-env/lib/python3.12/site-packages (from python-can==3.3.4->-r /home/szafran/klipper/scripts/klippy-requirements.txt (line 10)) (1.16.0)
Requirement already satisfied: aenum in ./klippy-env/lib/python3.12/site-packages (from python-can==3.3.4->-r /home/szafran/klipper/scripts/klippy-requirements.txt (line 10)) (3.1.15)
Collecting python-can==4.3
  Using cached python_can-4.3.0-py3-none-any.whl.metadata (8.7 kB)
Requirement already satisfied: wrapt~=1.10 in /home/szafran/klippy-env/lib/python3.12/site-packages (from python-can==4.3) (1.16.0)
Requirement already satisfied: packaging>=23.1 in /home/szafran/klippy-env/lib/python3.12/site-packages (from python-can==4.3) (24.1)
Requirement already satisfied: typing-extensions>=3.10.0.0 in /home/szafran/klippy-env/lib/python3.12/site-packages (from python-can==4.3) (4.12.2)
Requirement already satisfied: msgpack~=1.0.0 in /home/szafran/klippy-env/lib/python3.12/site-packages (from python-can==4.3) (1.0.8)
Using cached python_can-4.3.0-py3-none-any.whl (262 kB)
Installing collected packages: python-can
  Attempting uninstall: python-can
    Found existing installation: python-can 3.3.4
    Uninstalling python-can-3.3.4:
      Successfully uninstalled python-can-3.3.4
Successfully installed python-can-4.3.0
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ====
Wymagane jest uwierzytelnienie, aby uruchomić jednostkę „klipper-Mikrus.service”.
Authenticating as: Szafran (szafran)
Password:
==== AUTHENTICATION COMPLETE ====
```
and now I'm getting this error:
[image]

Don't really know what to do next, and would love some pointers on how to fix it.

dw-0, 2024-09-26:
Im sorry but i don't see why this could be a KIAUH issue.
Please head over to the Klipper Discord and try to get some support help there.

szafran81, 2024-09-26:
Sorry for that - I thought that maybe some of the scripts installs something wrong.
Anyways I'd like to leave this here for now (if You don't mind), and after I get some info on how to fix that I'll post here and close the Issue.

szafran81, 2024-09-26:
With the help of Sineos on klipper discourse I've managed to get klipper working.

This is what finally worked for me... maybe it'll help someone in the future:
```
sudo service klipper stop
~/klippy-env/bin/pip install -r ~/klipper/scripts/klippy-requirements.txt
cd ~/klippy-env/bin
./pip install --upgrade python-can~=4.3
./pip install --upgrade setuptools~=69.2.0
sudo service klipper start
```

closed 2024-09-26 by szafran81