dw-0/kiauh #547: python kiauh crashes right after launch

zamonary1, 2024-09-26:
### Linux Distribution

Armbian 24.5.1

### What happened

╔═══════════════════════════════════════════════════════╗
║ ~~~~~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~~~~~ ║
║        Klipper Installation And Update Helper         ║
║ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ║
╚═══════════════════════════════════════════════════════╝
[ERROR] An unexpected error occured:
int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Traceback (most recent call last):
  File "/home/zamonary1/kiauh/kiauh/core/menus/base_menu.py", line 184, in run
    self.display_menu()
  File "/home/zamonary1/kiauh/kiauh/core/menus/base_menu.py", line 178, in display_menu
    self.print_menu()
  File "/home/zamonary1/kiauh/kiauh/core/menus/main_menu.py", line 127, in print_menu
    self._fetch_status()
  File "/home/zamonary1/kiauh/kiauh/core/menus/main_menu.py", line 92, in _fetch_status
    self._get_component_status("mr", get_moonraker_status)
  File "/home/zamonary1/kiauh/kiauh/core/menus/main_menu.py", line 100, in _get_component_status
    status_data: ComponentStatus = status_fn(*args)
                                   ^^^^^^^^^^^^^^^^
  File "/home/zamonary1/kiauh/kiauh/components/moonraker/moonraker_utils.py", line 37, in get_moonraker_status
    return get_install_status(MOONRAKER_DIR, MOONRAKER_ENV_DIR, Moonraker)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zamonary1/kiauh/kiauh/utils/common.py", line 112, in get_install_status
    instances = len(get_instances(instance_type))
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zamonary1/kiauh/kiauh/utils/instance_utils.py", line 36, in get_instances
    instance_list = [
                    ^
  File "/home/zamonary1/kiauh/kiauh/utils/instance_utils.py", line 37, in <listcomp>
    instance_type(get_instance_suffix(name, service)) for service in service_list
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 7, in __init__
  File "/home/zamonary1/kiauh/kiauh/components/moonraker/moonraker.py", line 63, in __post_init__
    self.port: int | None = self._get_port()
                            ^^^^^^^^^^^^^^^^
  File "/home/zamonary1/kiauh/kiauh/components/moonraker/moonraker.py", line 144, in _get_port
    port: int | None = scp.getint("server", "port", fallback=None)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zamonary1/kiauh/kiauh/core/submodules/simple_config_parser/src/simple_config_parser/simple_config_parser.py", line 267, in getint
    return self._get_conv(section, option, int, fallback=fallback)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zamonary1/kiauh/kiauh/core/submodules/simple_config_parser/src/simple_config_parser/simple_config_parser.py", line 300, in _get_conv
    return conv(self.getval(section, option, fallback))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'

### What did you expect to happen

main menu of kiauh

### How to reproduce

launch kiauh.sh and select "yes" on prompt asking you to try python kiauh

dw-0, 2024-09-26:
Can you please upload your moonraker.conf file here?

zamonary1, 2024-09-26:
> Can you please upload your moonraker.conf file here?

Sure thing! 
[moonraker.zip](https://github.com/user-attachments/files/17153817/moonraker.zip)

dw-0, 2024-09-26:
Thank you. I see what the issue is. Due to the many includes, some expected config sections must be in different files. Thats why the error occurs. I'll prepare a fix for that.

closed 2024-09-26 by dw-0

zamonary1, 2024-09-27:
Totally works now. People like you are the future of FOSS. Thank you.