dw-0/kiauh #565: Error when installing Mainsail or Fluidd

Valdhor, 2024-10-12:
### Linux Distribution

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### What happened

```
╔═══════════════════════════════════════════════════════╗
║ ~~~~~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~~~~~ ║
║        Klipper Installation And Update Helper         ║
║ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ║
╚═══════════════════════════════════════════════════════╝
╔═══════════════════════════════════════════════════════╗
║ ~~~~~~~~~~~~~~~~~~~ [ Main Menu ] ~~~~~~~~~~~~~~~~~~~ ║
╟──────────────────┬────────────────────────────────────╢
║  0) [Log-Upload] │   Klipper: Installed: 1            ║
║                  │     Owner: Klipper3d               ║
║  1) [Install]    │      Repo: klipper                 ║
║  2) [Update]     ├────────────────────────────────────╢
║  3) [Remove]     │ Moonraker: Installed: 1            ║
║  4) [Advanced]   │     Owner: Arksine                 ║
║  5) [Backup]     │      Repo: moonraker               ║
║                  ├────────────────────────────────────╢
║  S) [Settings]   │        Mainsail: Not installed     ║
║                  │          Fluidd: Not installed     ║
║ Community:       │   Client-Config: Fluidd-Config     ║
║  E) [Extensions] │                                    ║
║                  │   KlipperScreen: Not installed     ║
║                  │       Crowsnest: Not installed     ║
╟──────────────────┼────────────────────────────────────╢
║  v6.0.0-alpha.6  │  Changelog: https://git.io/JnmlX   ║
╟──────────────────┴────────────────────────────────────╢
║                        Q) Quit                        ║
╚═══════════════════════════════════════════════════════╝
``` 
Select 1 (Install)
Select either 3 or 4 (Install Mainsail or Fluidd)

Get following error:

```
###### Perform action: 4
[ERROR] An unexpected error occured:
invalid literal for int() with base 10: '127.0.0.128126'
Traceback (most recent call last):
  File "/home/pi/kiauh/kiauh/core/menus/base_menu.py", line 188, in run
    selected_option.method(
  File "/home/pi/kiauh/kiauh/core/menus/install_menu.py", line 89, in install_fluidd
    client_setup.install_client(FluiddData())
  File "/home/pi/kiauh/kiauh/components/webui_client/client_setup.py", line 97, in install_client
    ports_in_use: List[int] = read_ports_from_nginx_configs()
  File "/home/pi/kiauh/kiauh/components/webui_client/client_utils.py", line 361, in read_ports_from_nginx_configs
    ports_to_ints_list = [int(port) for port in port_list]
  File "/home/pi/kiauh/kiauh/components/webui_client/client_utils.py", line 361, in <listcomp>
    ports_to_ints_list = [int(port) for port in port_list]
ValueError: invalid literal for int() with base 10: '127.0.0.128126'
``` 

### What did you expect to happen

I expected it to install the software

### How to reproduce

Using KIIAH install Klipper, then install Moonraker. Try to install either of the web interfaces.

### Additional information

This is on a Raspberry Pi 4B with 8GB RAM. Standard Raspbian install. As it's in a shed I have an external USB Wifi adapter (BrosTrend AX1800 USB WiFi 6 Linux Compatible WiFi Adapter) with their driver and the internal Wifi turned off.

I also have Octopi installed and working fine.

dw-0, 2024-10-13:
Hi, thanks for reporting.
You said you have octopi installed, so i assume you also use octoprint and serve its interface via nginx?
Can you provide the nginx configuration for it you have in `/ etc/ nginx/ sites-enabled`?
KIAUH tries to parse that file to check which port is already in use, so it can be prevented that two configs use the same port. It seems like the config KIAUH parsed had something unexpected in it, causing the parsing to fail.
If you could provide the content of the config (or the full config attached as a file to this issue) that would be nice.
Thanks in advance!

dw-0, 2024-10-13:
Yes, that might be it :)
[image]

closed 2024-10-13 by dw-0

Valdhor, 2024-10-13:
Perfect. Fixed. Thank you.

PR #568 (2024-10-13): fix: correctly handle IPs in nginx config files when parsing ports
Fixes #565
PR #570 (2024-10-13): Merge develop into master (v6.0.0-alpha.7)
fixes #561 
fixes #564 
fixes #565
commit 1b5691f2 (2024-10-13): Release v6.0.0-alpha.7

Merge develop into master (v6.0.0-alpha.7)

fixes #561
fixes #564
fixes #565