dw-0/kiauh #575: Trying to add Shake-Tune to multiple klippers installed with KIAUH

Bradford1040, 2024-10-14:
### Is your feature request related to a problem? Please describe

The system I am running this on is Debian Linux 13.0,Linux 6.11.0-061100-generic on x86_64, always using the newest version of KIAUH, think the newest one was V6, I also running Danger Klipper v0.12.0-480-gb2acb140, and have a total of 5 installs, but will change in a bit to 15, the number of them I don't think matters.

I am almost positive I got it figured out, but don't know enough about how KIAUH installs Klipper and virtual environments, I modified the original install script from shake-tune [this here](https://github.com/Frix-x/klippain-shaketune/blob/main/install.sh) I looked through some of the open and closed issues and found what I needed I think. [This Was First](https://github.com/Bradford1040/kiauh-klippain-shaketune/blob/BETA_V1/install.sh) attempt and I have not done anything to the install or config files beside add the shake-tune update stuff to moonraker.cfg and some options in printer.cfg

A friend on ZeroG Discord server helped edit [install.sh](https://github.com/Bradford1040/kiauh-klippain-shaketune/blob/BETA_V2/install.sh) I have not used it yet. I figured I would ask for more confirmation from the KIAUH devs  

### Describe the solution you'd like

To be able to just install like everything else I have install with KIAUH, which truly is a remarkable program. The Klipper experience has most likely been much easier thanks to this program! 

### Describe alternatives you've considered

Tried installing myself as described above

dw-0, 2024-10-14:
Hi, im not really familiar with the klippain / shake-tune project but i had a brief glance over the install script.

No matter how many klipper instances there are, KIAUH sets everything up so they all uses a single klipper and klipper-env folder. That means, the script seems to be fine in regards to installing dependencies to the klipper env and linking the module to the klippy/extras folder.

Some steps need to be looped for all instances though:
- https://github.com/Bradford1040/kiauh-klippain-shaketune/blob/BETA_V2/install.sh#L103
   - this part seems to remove a folder from the printer_data dir, that means this needs to be done for all printer_data folder with their respective names
- https://github.com/Bradford1040/kiauh-klippain-shaketune/blob/BETA_V2/install.sh#L128
   - Here you need to modify the moonraker.conf of each instance, residing in each instances printer_data folder 
- https://github.com/Bradford1040/kiauh-klippain-shaketune/blob/BETA_V2/install.sh#L136
   - here you need to restart all affected klipper services
- https://github.com/Bradford1040/kiauh-klippain-shaketune/blob/BETA_V2/install.sh#L141
   - here you need to restart all affected moonraker services 

Those functions need to be modified, so that they can handle multiple klipper services and the different printer_data folders.
I think thats it to make it at least work...somehow :)

Bradford1040, 2024-10-14:
The script that that was with the original before I modified it, could not install at all as when you use custom name installing with KIAUH I guess it does not make a klipper.service which his install.sh looks for, the one I included in my first message that I modifed was the only way I was able to get it to install, and I was just pleading kindly for you to look and see if I done anything wrong (FROM WHAT YOU CAN TELL) I am not holding any iron on your feet lol, just trying to see if my old butt modified correctly 

[THIS IS MY MOD](https://github.com/Bradford1040/kiauh-klippain-shaketune/blob/BETA_V1/install.sh), if I did okay and the above info is the only adds you can think of, I am pretty sure you can close the ticket

Bradford1040, 2024-10-14:
also on it's own, just wanted to thank you for your amazing hard work, KIAUH is maybe one of the best done apps I have seen as an end user, THANK YOU

Bradford1040, 2024-10-15:
I would feel this is my last edit, and sorry if asking you for this help is out of line, The entire Danger Klipper discord server told me to uninstall KIAUH and install klipper the normal way, and trust when I say I gave them the KIAUH does install klipper normally lol. But also the room I was in acted like I had just discovered fire when I told them I was running multiple machines off one Server/PC/LapTop & or Rpi. I did include links to your GitHub and explained how easy it was and versatile, including installing custom repo's like Danger Klipper.

Now almost every community dealing with DYI or Kit CoreXY printers (I myself converted my Ender 5 Plus to ZeroG Mercury one.1, the Voron guys, VZ_Bot) pretty much everyone has told me to get this Klippain-shaketune, it would seem as a new person to this community that looking into adding this to you custom extra's would be benefiting your end users.

     https://github.com/Bradford1040/kiauh-klippain-shaketune/blob/BETA_V3/install.sh 

This is a edit I feel would solve the install issues and trust me the ORIGINAL > [install.sh](https://github.com/Bradford1040/kiauh-klippain-shaketune/blob/main/install.sh) < Will Not Install on to a fresh install of Klipper with using KIAUH, I tried it, only way it did work, The original install.sh worked was if you only installed one printer NOT Multiple, regardless if you name with your own custom names or even with default multiple naming example = "printer_1_data"

dw-0, 2024-10-17:
> The entire Danger Klipper discord server told me to uninstall KIAUH and install klipper the normal way, and trust when I say I gave them the KIAUH does install klipper normally lol.

There are always people who prefer different ways of doing things to achieve a specific goal. There might be no point in arguing then. But you are correct, the typical 1 instance install is the regular, normal install we all know. It's just with multi instance, that KIAUH does special things. If someone doesn't like that, projects like prind by mkuf might be more suitable and people have to spin up as many docker container as they like.

> everyone has told me to get this Klippain-shaketune, it would seem as a new person to this community that looking into adding this to you custom extra's would be benefiting your end users

Im open for contributions of new community extensions for KIAUH v6. For v6 i made it (what i think and feel) much easier to install klipper extras of any kind. Klippain was actually the reason (or one of the reasons) i developed that new "community extension" feature for KIAUH.

> The original install.sh worked was if you only installed one printer NOT Multiple

Sorry if this wasn't clear from my explanation above, but yes... absolutely the original install.sh will NOT work with multiple instances. Hence my pointers to the parts of the script that has to be modified to likely make it work.

Bradford1040, 2024-10-25:
I am extremely happy with your help and willingness to explain an out of the normal fix, I also would like to add that every chance I have I give praise to you and KIAUH, in all honesty, without KIAUH I would not have the set up I do have, I was on Marlin before I came I think it was Chris's basement youtube video of KIAUH, and at first was talking about installing multiple Klipper on a Rpi, But I was looking at installing 6 printers (I now have 9 almost 10) and camera's on each, the Rpi 3b I had sounded like it needed steroids if it was to even have a shot of just the 6 printers, so did some more searching and found PC install stuff. I use an old Dell poweredge R720XD and it runs extremely smooth (no little wait circles lol) 

Basically I wanted you to understand I am so happy I found your project and also wanted you to know how much your project has helped people and myself, I was almost destitute after I lost my manufacturing business from the Covid shutdown, the printers gave me a new revenue so I was able to get back into a house and put food in my belly, I believe your work has done that for me. so Thank you

Bradford1040, 2024-10-25:
[This is my fix](https://github.com/Bradford1040/kiauh-klippain-shaketune/blob/BETA_V3/install.sh)

this is a fix that will only install for one of the custom name printers installed via KIAUH, I do not know regex stuff to seach out each service that is created but if you have installed multiple klipper installs and have just gone with the standard KIAUH naming scheme then `punisher` (the name of one of my printers) needs to be changed with `printer_1` or the folder/printer you want it installed to. I hope that makes sense to future readers. 

The script absolutely works as I have tried it several times recently do to changing hardware around.

dw-0, 2024-10-25:
Hi, im glad it worked out for you!
If you still want to tweak it and make it work with multi instances, here are maybe some pointers for you:
This is how KIAUH detects all available klipper services: https://github.com/dw-0/kiauh/blob/master/scripts/klipper.sh#L25
This is how you can get only the name portion of that custom named multi instance: https://github.com/dw-0/kiauh/blob/master/scripts/utilities.sh#L619
Feel free to use and copy the code as needed.

closed 2024-10-31 by dw-0