dw-0/kiauh #579: interface error from installing spoolman

sastreae, 2024-10-18:
### Linux Distribution

debian 11

### What happened

at the time of installing spoolman. A bad command line is added to the file /home/pi/printer_data/moonraker.asvc
. The word spoolman is not written on a new line but is written next to the last installation

### What did you expect to happen

It should be written one line below so that it is well read by the system

### How to reproduce

first installation of klipper, installing several more programs, when installing spoolman an error appears on the screen

### Additional information

When the error appears, it does not allow you to give screen commands. but continue accessing normally via ssh.

dw-0, 2024-10-20:
Thanks for the report. I was able to reproduce the issue and will provide a fix soon.

sastreae, 2024-10-21:
thanks to you.

PR #581 (2024-10-20): fix(KIAUH v5): correctly add Spoolman to moonraker.asvc
Fixes #579
PR #583 (2024-10-21): Merge develop into master (Release v6.0.0-alpha.8)
fixes #571 
fixes #579
closed 2024-10-21 by dw-0