dw-0/kiauh #589: New version V6 issue

Bradford1040, 2024-10-23:
### Linux Distribution

Linux 6.11.0-061100-generic on x86_64
Debian Linux 13.0 (aka Rhino Linux)
Running Danger Klipper version = v0.12.0-519-gc8acb287

### What happened

I had switched to V6 about a week ago and last night had a MCU error during a print, think it was accelerator LIS2DW I had forgotten to set `is_non_critical: True` under the extra mcu, but took the printers being down at this time to upgrade Danger Klipper Master to Bleeding_Edge_V2 and used KIAUH V6 to try and get this done, I had many dependency errors!

[DK_ERROR.txt](https://github.com/user-attachments/files/17499057/DK_ERROR.txt)
Copy of the error via copied from cmd line.

### What did you expect to happen

danger klipper bleeding edge branch to be installed same as master branch was

### How to reproduce

Install Main Line klipper, then install Danger Klipper Master branch via KIAUH V5 then try to switch branch with KIAUH now V6 to bleeding_edge_v2 branch

dw-0, 2024-10-24:
Hi, looking at the issue i cannot see an issue directly related to KIAUH. There is an issue during the installation of the dependencies. Those are grabbed from the provided install scripts and requirements files provided in the danger klipper repository. I was able to switch to the branch without any issue on a Ubuntu 22 VM.

Bradford1040, 2024-10-25:
I think it was just me somehow, as I have spoken to others that have installed danger klipper with your V6 version, but that was the error I had gotten, so figured I would bring it to your attention.

closed 2024-10-25 by dw-0