dw-0/kiauh #591: Installing and uninstalling KlipperScreen, System boots in Terminal mode

giubu, 2024-10-25:
### Linux Distribution

MX-23
Liquorix Kernel
amdgpu

### What happened

After installing and uninstalling KlipperScreen,, the system boots in terminal mode. I can log in and `startx` , which takes me to the GUI, after which I can use the system normally. Qjackctl was also uninstalled in the process.

### What did you expect to happen

After uninstall, use the system as per normal

### How to reproduce

install KlipperScreen, reboot, uninstall KlipperScreen

dw-0, 2024-10-25:
Hi, from the provided information i can not directly see an issue regarding KIAUH.
The KlipperScreen installation provided by KIAUH basically calls the official KlippeScreen installation script.

"Uninstalling KlipperScreen" with KIAUH does the following:
- Deleting its service
- Deleting its source folder
- Deleting its virtual environment folder
- Deleting its logfiles

No dependencies or system packages are uninstalled in that process. So if `Qjackctl` is missing (i don't know what this software does), it is not due to calling the KlipperScreen uninstaller of KIAUH.

Overall i don't think this is an issue i can help with troubleshoot. Im not even sure if it something @alfrix could help with. For me, personally, there is not enough information. Apart from that im not experienced enough with KlipperScreen.

giubu, 2024-10-25:
Thank you for your reply.
I think the problem might be more related to the installation script rather than the uninstall one. I am not an expert and I might have done something fundamentally wrong, but as far as I can tell I only installed it, rebooted the system, uninstalled it. Something must have happened during this process. Qjackctl is the main audio connection application in many Linux systems
[https://qjackctl.sourceforge.io/](url)
basically, the installation of KlipperScreen interfered with the previous audio and video setup, as both the video and sound managers of the system have been compromised.

dw-0, 2024-10-25:
Okay i see. If you think that something went south during the installation step, i would suggest you head over to the KlipperScreen repository and open an issue there, explaining the problem once again. Maybe the maintainer of KlipperScreen can give more insight of what has happened or why you found yourself in the current state after uninstallation. Sorry that i cannot help you any further.

alfrix, 2024-10-25:
hi, i'm reading this due to the ping, the only thing that could make it boot to terminal after an uninstall is an install that was set as a service, it literally tells you that it will enable boot to console:
```
"It will create a service, enable boot to console and install the graphical dependencies."
```
to revert it you would need to do:
```
sudo systemctl set-default graphical.target && sudo reboot
```
nothing to do with qjackctl, that makes no sense to me

dw-0, 2024-10-25:
Thank you for chiming in @alfrix. @giubu please try the suggested solution and see if it works.

closed 2024-10-31 by dw-0