dw-0/kiauh #593: Adding client config failing on new install?

novaplusplus, 2024-11-06:
### Linux Distribution

Raspberry Pi OS lite, 32-bit

### What happened

On a brand new install, KIAUH v6 reported "[INFO] Another Client-Config is already installed! Skipped ..." when attempting to install the fluidd config file.

### What did you expect to happen

Being allowed to install it when there's nothing there :shrug: 

### How to reproduce

Probably doing a fresh install and then add fluidd-config, haven't tried mainsail. I've done it twice so far with fluidd though (clean reinstall)

### Additional information

I managed to make it work by simply commenting out the check here:
https://github.com/dw-0/kiauh/blob/a63cf8c9d911d0a068dd126754dd64665df821ae/kiauh/components/webui_client/client_config/client_config_setup.py#L41

Then it properly loaded the repository and added it to moonraker.conf.

It's very possible I did something wrong when doing it the frist time but I'm not sure what step might have done so.

Stemolap, 2024-11-06:
Debian 11 Bullseye Image for the SonicPad, freshly installed and updated. Using kiauh v6 generates same error both on fluidd and mainsail.
[image]

dw-0, 2024-11-10:
@Stemolap could it be, that both folders for the mainsail-config and fluidd-config exist, but your printer.cfg does not contain a `[include mainsail.cfg]` or `[include fluidd.cfg]` section?

MaroonOut09, 2024-11-18:
I'm having the same issue with a fresh install on Raspberry Pi OS Lite 64-bit (Bookworm).

I tried adding `[include mainsail.cfg]` and `[include fluidd.cfg]`, and it did not resolve the issue.

closed 2025-03-15 by dw-0