dw-0/kiauh #594: Make a headless mode

jhi2, 2024-11-06:
### Is your feature request related to a problem? Please describe

No, it is a request.

### Describe the solution you'd like

I want to run like:
```shell
./kiauh.sh 1 7
``` 
to install klipperscreen, instead of running:
```shell
./kiauh.sh
Select an option:1
Select an option:7
```

jhi2, 2024-11-11:
@dw-0

dw-0, 2024-11-11:
@jhi2 i saw your request. No need to ping me again. What do you expect? That this feature gets instantly implemented?

jhi2, 2024-11-11:
> @jhi2 i saw your request. No need to ping me again. What do you expect? That this feature gets instantly implemented?

Sorry.
Thank you anyways!

jhi2, 2024-11-17:
> @jhi2 i wrote a script doing that a while ago. Just created a fork to push it to, will post a link if ready.

THANKS!

igem0n, 2025-02-26:
I'm also interested in some way of an automated deploy. for me passing exact services names looks better than just numbers, but any way to make it in non-interactive manner would be great!

igem0n, 2025-03-07:
I've just realized that it would be nice to provide a bit more context on it. 
My main goal is to have some kind of full backup/restore functionality with not only config files, but with services as well.
As long as we already have nicely working configs backup service, I assume kiauh could create its own config file like **kiauh.cfg** in the printer data folder with a listing of the services it installed and parameters used for installation.
Later for the restore stage, we could just call kind of `./kiauh/kiauh.sh --restore https://github.com/myself/my-config-repo.git` to clone backed up configs and if **kiauh.cfg** is present there, use it to restore services previously being installed.
Or just clone configs repo manually and then point kiauh to the local folder where all the configs with  **kiauh.cfg** are stored, like `./kiauh/kiauh.sh --restore ~/printer_data/configs`

sqanch, 2025-03-23:
You can achieve the behavior desired by @jhi2 with the following command:
`echo -e "1\n7" | ./kiauh.sh`

WTMike24, 2025-05-10:
Adding here - if you want to preview the menus before running the script headless you can find the install menu paths here:

First menu: Version 5 or Version 6 menu (can be skipped on future runs if you select one of the remember-me options)
https://github.com/dw-0/kiauh/blob/6f59fd06aad4ea78104cb71dccb82ea7a2b1878d/kiauh.sh#L137

Below are the v6 menu options which worked just fine for me

Main menu:
https://github.com/dw-0/kiauh/blob/6f59fd06aad4ea78104cb71dccb82ea7a2b1878d/kiauh/core/menus/main_menu.py#L131

Install menu:
https://github.com/dw-0/kiauh/blob/6f59fd06aad4ea78104cb71dccb82ea7a2b1878d/kiauh/core/menus/install_menu.py#L62

You may have to do some additional digging if you prefer v5 or if you want to follow other menu paths.