dw-0/kiauh #595: kiauh v6 python dependency install failures

eriknz, 2024-11-09:
### Linux Distribution

Raspberry Pi OS Lite (32-bit)

### What happened

I'm trying to set up klipper via kiauh on a couple of raspberry pi's (a 2 and a 3). First go installing klipper on the pi 2, I got the following errors:
```
###### Installing Python requirements ...
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/pyserial/
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/markupsafe/
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/wrapt/
```
That was using kiauh v6. I tried removing klipper then reinstalling it using kiauh v5 and it worked. 
I tried doing a new setup on the pi 3 (same OS = Raspberry Pi OS Lite (32-bit)) and using kiauh v6 this time it got through klipper installation ok but then had python dependency errors during Moonraker installation:
```
###### Installing Python requirements ...
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/pillow/
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/preprocess-cancellation/
```
Again removing it and reinstalling using kiauh v5 fixed it.

### What did you expect to happen

it to install without errors

### How to reproduce

fresh install on raspberry pi 2 or 3 using Raspberry Pi OS Lite (32-bit) and kiauh v6.

dw-0, 2024-11-09:
Please make sure you have a stable internet connection. Those error messages indicate a network issue. Although it worked on the attempts with v5, I think that was more of a coincidence. As you stated yourself, Klipper installed fined with v6 on the Pi 3.

eriknz, 2024-11-09:
Ok. As far as I know my internet connection is good, I can get over 90Mbps up and down (I'm on fibre). Wifi signal strength on the pi 2 was a bit low at -62dBm (in the printer a couple rooms away) but -29dBm on the pi 3 that is right near the router.

dw-0, 2024-11-09:
Im rather sure if you try to reproduce, but this time use cable instead of wifi, it would work pretty fine without any issues.
I can just relate to the warning messages you provided, and they clearly indicate some kind of network issue. Might not be on your end, and somewhere in between you and the servers providing the packages to download.

closed 2024-11-10 by dw-0

slothking87, 2025-05-03:
Just to add to this - I experienced the same error, Erased the new Kiauh folder, cloned V5 and install worked fine. I tried install with V6 multiple times.

marine-ecologist, 2025-05-30:
Same error with KIAUH v6 on a fresh install of OS 32 lite, both with cable and wifi: 

```
Created wheel for MarkupSafe: filename-markupsafe-3.0.2-cp311-cp311-11nux_armu71.uhl size=2190? sha256=d5904e4c4041101f19e1b61
31ff48f30995599fe0855fb7d8b57b6354fb9c2ac
Stored in directory: /home/admin/.cache/pip/wheels/9đ/38/99/1f61f3b0dđ7ab4898edfa9fcf6feb13644d4đ49a44b3bed19đ
Successfully built MarkupSafe
Installing collected packages: python-periphery, pyserial, paho-ngtt, Ifaddr, zipp, zeroconf, vrapt, urllib3, tornado, PyYAML, P yasn1, preprocess-cancellation, pillou, oauthlib. MarkupSafe, markdoun, libnacl, Inotify simple, idna, distro, abus-fast, click, charset-normalizer, certifi, snart-open, requests, ldap3, jinja2, importlib_metadata, streaming-form-data, requests-oauthlib, a
Successfully installed MarkupSafe-3.0.2 PyYAML-6.0.2 apprise-1.9.2 certifi-2025.4.26 charset-normalizer-3.4.2 click-8.2.1 dbus-f ast-2.28.0 distro-1.9.0 idna-3.10 ifaddr-0.2.0 importlib_metadata-8.2.0 inotify simple-1.3.5 Jinja2-3.1.5 1dap3-2.9.1 libnacl-2.
1.0 markdown-3.8 oauthlib-3.2.2 paho-ngtt-1.6.1 pillow-11.1.0 preprocess-cancellation-0.2.1 pyasn1-0.6.1 pyserial-3.4 python per iphery-2.4.1 requests-2.32.3 requests-oauthlib-2.0.0 snart-open-?.1.0 streaning-forn-data-1.19.1 tornado-6.4.2 ur 11ib3-2.4.0 ura
pt-1.17.2 zeroconf-0.131.0 zipp-3.22.0
WARNING: Retrying (Retry(total=4, connect-None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolErra r('Connection aborted.', RenoteDisconnected(' Remote end closed connection without response' ))': /simple/charset-normalizer/
CERRORI
installing Python requirenents: Installing Python requirements failed!
CERRORI
Error during installation of Moonraker requirements?
LERRORI
Error while installing Moonraker: Error installing Python requirements: Installing Python requirements failed!
```

Reinstalling Moonraker uses all cached packages, but still gives the same python dependency error during installation even though packages are already cached and there's no connection to abort.

If I go through the reinstall of Moonraker after this error (create example config, don't overwrite existing repository, don't re-create Virtualenv) then Moonraker installs successfully with:

```
###### Start moonraker.service ...
[OK] OK!

```

DeanCording, 2025-05-31:
I am experiencing this issue as well.  My investigations suggest that the problem is that pip returns a non-zero exit code when it logs a warning message (such as needing to retry downloading a package).  In version 6, the code fails if the exit code is not zero, whereas in version 5 it does not check the exit code.

DeanCording, 2025-05-31:
Further testing shows this output:

 ```
 WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/pillow/pillow-11.1.0-cp311-cp311-linux_armv7l.whl
  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/zeroconf/zeroconf-0.131.0-cp311-cp311-manylinux_2_36_armv7l.whl
  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /simple/dbus-fast/dbus_fast-2.28.0-cp311-cp311-manylinux_2_36_armv7l.whl

Return code 0
[ERROR] Error installing Python requirements: Installing Python requirements failed!
[ERROR] Error during installation of Moonraker requirements!
[ERROR] Error while installing Moonraker: Error installing Python requirements: Installing Python requirements failed!
```

So, even though the result.returncode is 0, the installation procedure fails because there are warning messages of result.stderr.  It should not fail if result.returncode is 0.

Suggest changing `if result.returncode != 0 or result.stderr:` to `if result.returncode != 0:`