dw-0/kiauh #597: Kiauh does not recognize Python 3.11

manuelchaves, 2024-11-10:
### Linux Distribution

Armbian 24.8.1 Bookworm Minimal 

### What happened

Donf find Python 3.11

### What did you expect to happen

Old version (5.x) its OK

### How to reproduce

Upgrade it 

### Additional information

[image]

dw-0, 2024-11-11:
Hi, thanks for the report. Can you please post the output of `python3 -V` ?
That is the command KIAUH uses to determine the actual python 3 version. But tbh, it would be odd if that would show an older python 3 version like the one shown with `python -V`.

manuelchaves, 2024-11-11:
Tjanks
mks@mkspi:~$ python3 -V
Python 3.11.0
mks@mkspi:~$

L8nitesurfer, 2024-11-12:
So I ran into this issue and tried the suggestion above but I am still encoutering the error. 
ΓòöΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòù
Γòæ ~~~~~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~~~~~ Γòæ
Γòæ        Klipper Installation And Update Helper         Γòæ
Γòæ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ Γòæ
ΓòÜΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓòÉΓò¥
[ERROR] An unexpected error occured:
Expecting value: line 1 column 1 (char 0)
Traceback (most recent call last):

Additionally I removed the folder and its content and reinstalled but the issue presist.

dw-0, 2024-11-12:
>  tried the suggestion above

There is no suggestion anywhere.

Im also not sure what kind of error you get here and i don't think it is related to this topic.

dw-0, 2024-11-12:
> Thanks mks@mkspi:~$ python3 -V Python 3.11.0 mks@mkspi:~$

Interesting. So `python -V` and `python3 -V` are both available and both show the same output. I don't really understand why that error is thrown. I somehow need to figure out how to reproduce this issue.

closed 2025-02-09 by dw-0