dw-0/kiauh #598: (v6.0.0-alpha.9) Attempting to update an up-to-date package returns wrong error.

HumbleDeer, 2024-11-12:
### Linux Distribution

Debian 12.8 w/o UI ("standard")

### What happened

When accessing the Update menu, trying to select a package for updating when it is already up-to-date properly assesses it doesn't need updating, throws an error to the handler, but returns the wrong error message.

### What did you expect to happen

The error message should reflect the actual situation by informing the user that the action was skipped because the package was already up-to-date.

### How to reproduce

1. Access the Update menu
2. Select a package for update that is already up-to-date

### Additional information

[image]

HumbleDeer, 2024-11-24:
Hi! Crowsnest was fully installed and was not shown as incomplete. It was, however, installed by the previous version of KIAUH. It was not installed by the beta version. I currently can't reproduce it anymore either as I ended up removing crowsnest, not needing it anymore.

dw-0, 2024-11-24:
Thank you for the feedback. Im not sure if the installation with KIAUH v5 could cause this.
Luckily this is not a serious issue, so for now i would close this issue. But if you encounter it once more, please report it again by either posting it into this issue or open a new one, so i can have a closer look again.

closed 2024-11-24 by dw-0