dw-0/kiauh #605: Cant install Crowsnest

LXZ993, 2024-12-05:
### Linux Distribution

Debian Bookworm (Raspberry Pi OS Lite (32-bit)

### What happened

`[ 90%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/pollinterrupter.cpp.o
[ 92%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/pollservice.cpp.o
[ 92%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/tcpserver.cpp.o
[ 92%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/tcptransport.cpp.o
[ 95%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/tlstransport.cpp.o
[ 95%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/verifiedtlstransport.cpp.o
[ 95%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/websocket.cpp.o
/home/LXZ/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/tlstransport.cpp: In constructor ‘rtc::impl::TlsTransport::TlsTransport(std::shared_ptr<rtc::impl::TcpTransport>, std::optional<std::__cxx11::basic_string<char> >, rtc::impl::certificate_ptr, rtc::impl::Transport::state_callback)’:
/home/LXZ/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/tlstransport.cpp:351:58: warning: ‘void EC_KEY_free(EC_KEY*)’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  351 |                 auto ecdh = unique_ptr<EC_KEY, decltype(&EC_KEY_free)>(
      |                                                          ^~~~~~~~~~~
In file included from /usr/include/openssl/x509.h:33,
                 from /usr/include/openssl/ssl.h:31,
                 from /home/LXZ/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/tls.hpp:56,
                 from /home/LXZ/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/certificate.hpp:24,
                 from /home/LXZ/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/tlstransport.hpp:22,
                 from /home/LXZ/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/tlstransport.cpp:19:
/usr/include/openssl/ec.h:1003:28: note: declared here
 1003 | OSSL_DEPRECATEDIN_3_0 void EC_KEY_free(EC_KEY *key);
      |                            ^~~~~~~~~~~
/home/LXZ/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/tlstransport.cpp:352:45: warning: ‘EC_KEY* EC_KEY_new_by_curve_name(int)’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  352 |                     EC_KEY_new_by_curve_name(NID_X9_62_prime256v1), EC_KEY_free);
      |                     ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/usr/include/openssl/ec.h:998:31: note: declared here
  998 | OSSL_DEPRECATEDIN_3_0 EC_KEY *EC_KEY_new_by_curve_name(int nid);
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~
/home/LXZ/crowsnest/bin/camera-streamer/third_party/libdatachannel/src/impl/tlstransport.cpp:352:80: warning: ‘void EC_KEY_free(EC_KEY*)’ is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  352 |                     EC_KEY_new_by_curve_name(NID_X9_62_prime256v1), EC_KEY_free);
      |                                                                                ^
/usr/include/openssl/ec.h:1003:28: note: declared here
 1003 | OSSL_DEPRECATEDIN_3_0 void EC_KEY_free(EC_KEY *key);
      |                            ^~~~~~~~~~~
[ 97%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/websocketserver.cpp.o
[ 97%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/wstransport.cpp.o
[ 97%] Building CXX object CMakeFiles/datachannel-static.dir/src/impl/wshandshake.cpp.o
[100%] Linking CXX static library libdatachannel-static.a
make[5]: Leaving directory '/home/LXZ/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
[100%] Built target datachannel-static
make[4]: Leaving directory '/home/LXZ/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
make[3]: Leaving directory '/home/LXZ/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
make[2]: Leaving directory '/home/LXZ/crowsnest/bin/camera-streamer/third_party/libdatachannel/build'
make[1]: Leaving directory '/home/LXZ/crowsnest/bin/camera-streamer'
make: *** [Makefile:67: all] Error 2
make: *** [Makefile:19: install] Error 2
[ERROR] Something went wrong! Please try again...
Command 'sudo make install BASE_USER=LXZ' returned non-zero exit status 2.
`

### What did you expect to happen

IDK?

### How to reproduce

Fresh install of Klipper + Moonraker + Mainsail

### Additional information

Have tried to reinstall the system multiple times. No look i get allways the same error.

PrimAvalou, 2024-12-05:
Happy to have an solution for this too, I have exactly the same error

dw-0, 2024-12-07:
It looks like there is currently a compilation error for camera-streamer (which Crowsnest uses) on Bookworm.
This is no actual issue with KIAUH, nor with Crowsnest. The root issue lies in camera-streamer.
Im going to close this issue. You sadly need to wait for a fix to the camera-streamer project.

closed 2024-12-07 by dw-0