dw-0/kiauh #606: v6 bugs: Kiauh Klipper Install constantly fails in v6

jjkark, 2024-12-08:
### Linux Distribution

6.6.62+rpt-rpi-v7

### What happened

Klipper fails to install, crashing on "Receiving objects" with a error message of:

error: RPC failed; curl 56 GnuTLS recv error (-110): The TLS connection was non-properly terminated.
error: 3719 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
[ERROR] Error cloning repository https://github.com/Klipper3d/klipper: Unknown error
[ERROR] An unexpected error occured during cloning of the repository.
[ERROR] An unexpected error occured during cloning of the repository.
[ERROR] Klipper installation failed!

### What did you expect to happen

I expected to have KIAUH install Klipper, but it fails 100% of the time.

### How to reproduce

even with expanding the http and https buffers to 500mb, it does not resolve the issue. I can reproduce 100% of the time.
[image]

### Additional information

[image]
[image]

jjkark, 2024-12-08:
[image]
b

jjkark, 2024-12-08:
[image]

Even tried changing speed and timeouts:

pi@raspberrypi:~ $ git config --global http.postBuffer 524288000
pi@raspberrypi:~ $ git config --global http.lowSpeedLimit 0
pi@raspberrypi:~ $ git config --global http.lowSpeedTime 999999
pi@raspberrypi:~ $ git config --global https.postBuffer 524288000

dw-0, 2024-12-09:
Looking at the screenshots, your download speed is very very slow. This could lead to such issues as well.
Can you successfully install Klipper with KIAUH v5?

jjkark, 2024-12-09:
Yes. My download speed on v5 was 1.2mb/s up to 1.8mb/s. Only on v6 was it throttled. 

This was v5

Receiving objects:  13% (9698/69390), 62.57 MiB | 1.02 MiB/s
Receiving objects:  54% (37471/69390), 185.85 MiB | 838.00 KiB/s 
Receiving objects:  87% (57147/69390), 312 MiB | 1.14 MiB/s

this was v6:

[image]

jjkark, 2024-12-09:
Whoops. Followed by these two. 

[image]
[image]

dw-0, 2024-12-09:
Can you please do me a favor and show me the installation process via actual screenshots of Klipper with KIAUH v5?
Im unsure currently what's going on there, i don't think there should be a difference, as both v5 and v6 use the same clone command (i think). So current screenshots of v5 would be a very good hint if its possible for you.

jjkark, 2024-12-09:
As it is already fully installed now, what would you like me to do? Remove and reinstall via v5?

dw-0, 2024-12-09:
> Remove and reinstall via v5?

Ideally, yes.
But how did you manage to install it with v6 now? I think it did not work at all?

zubairahmed01, 2024-12-22:
I am facing the similar error weither it is >V6 or V5, I am getting

´error: 8099 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output´

I was also not able to clone kilpper initially if I do it manually, but after changing the git http post buffer, I can clone but the script still fails.

https://stackoverflow.com/questions/6842687/the-remote-end-hung-up-unexpectedly-while-git-cloning

 'git config --global http.postBuffer 1048576000'

sreknob, 2024-12-25:
Just including that I got the same error, but seems more like a git failure and probably not KIAUH related.
Just ran the installation a second time and cloning was successful.

```
/=======================================================\
|         KIAUH v6.0.0-alpha1 is available now!         |
|-------------------------------------------------------|
|         View Changelog: https://git.io/JnmlX          |
|                                                       |
| KIAUH v6 was completely rewritten from the ground up. |
| It's based on Python 3.8 and has many improvements.   |
|                                                       |
| NOTE: Version 6 is still in alpha, so bugs may occur! |
| Yet, your feedback and bug reports are very much      |
| appreciated and will help finalize the release.       |
|-------------------------------------------------------|
| Would you like to try out KIAUH v6?                   |
| 1) Yes                                                |
| 2) No                                                 |
| 3) Yes, remember my choice for next time              |
| 4) No, remember my choice for next time               |
|-------------------------------------------------------|
|                        Q) Quit                        |
\=======================================================/
###### Select action: 1
╔═══════════════════════════════════════════════════════╗
║ ~~~~~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~~~~~ ║
║        Klipper Installation And Update Helper         ║
║ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ║
╚═══════════════════════════════════════════════════════╝
╔═══════════════════════════════════════════════════════╗
║ ~~~~~~~~~~~~~~~~~~~ [ Main Menu ] ~~~~~~~~~~~~~~~~~~~ ║
╟──────────────────┬────────────────────────────────────╢
║  0) [Log-Upload] │   Klipper: Not installed           ║
║                  │     Owner: -                       ║
║  1) [Install]    │      Repo: -                       ║
║  2) [Update]     ├────────────────────────────────────╢
║  3) [Remove]     │ Moonraker: Not installed           ║
║  4) [Advanced]   │     Owner: -                       ║
║  5) [Backup]     │      Repo: -                       ║
║                  ├────────────────────────────────────╢
║  S) [Settings]   │        Mainsail: Not installed     ║
║                  │          Fluidd: Not installed     ║
║ Community:       │   Client-Config: -                 ║
║  E) [Extensions] │                                    ║
║                  │   KlipperScreen: Not installed     ║
║                  │       Crowsnest: Not installed     ║
╟──────────────────┼────────────────────────────────────╢
║ v6.0.0-alpha.12  │  Changelog: https://git.io/JnmlX   ║
╟──────────────────┴────────────────────────────────────╢
║                        Q) Quit                        ║
╚═══════════════════════════════════════════════════════╝
###### Perform action: 1
╔═══════════════════════════════════════════════════════╗
║ ~~~~~~~~~~~~~~~ [ Installation Menu ] ~~~~~~~~~~~~~~~ ║
╟───────────────────────────┬───────────────────────────╢
║ Firmware & API:           │ Touchscreen GUI:          ║
║  1) [Klipper]             │  7) [KlipperScreen]       ║
║  2) [Moonraker]           │                           ║
║                           │ Webcam Streamer:          ║
║ Webinterface:             │  8) [Crowsnest]           ║
║  3) [Mainsail]            │                           ║
║  4) [Fluidd]              │                           ║
║                           │                           ║
║ Client-Config:            │                           ║
║  5) [Mainsail-Config]     │                           ║
║  6) [Fluidd-Config]       │                           ║
╟───────────────────────────┴───────────────────────────╢
║                       B) « Back                       ║
╚═══════════════════════════════════════════════════════╝
[... 52 lines cut ...]
║ up. The number of Klipper instances will determine    ║
║ the amount of printers you can run from this host.    ║
║                                                       ║
║ WARNING:                                              ║
║ Setting up too many instances may crash your system.  ║
╟───────────────────────────────────────────────────────╢
║                       B) « Back                       ║
╚═══════════════════════════════════════════════════════╝
###### Number of Klipper instances to set up (default=1):
###### Create example printer.cfg? (Y/n): y

###### Cloning repository from 'https://github.com/Klipper3d/klipper'
Cloning into '/home/blake/klipper'...
remote: Enumerating objects: 69869, done.
remote: Counting objects: 100% (78/78), done.
remote: Compressing objects: 100% (52/52), done.
remote: Total 69869 (delta 63), reused 26 (delta 26), pack-reused 69791 (from 3)
Receiving objects: 100% (69869/69869), 197.39 MiB | 3.69 MiB/s, done.
Resolving deltas: 100% (56442/56442), done.
Updating files: 100% (2275/2275), done.
[OK] Clone successful!
Already on 'master'
Your branch is up to date with 'origin/master'.
[OK] Checkout successful!
```

dw-0, 2024-12-26:
As the reported issues rather point to a network related issues, i will close this topic. I cannot see an issue specific to KIAUH.

closed 2024-12-26 by dw-0

marine-ecologist, 2025-05-30:
> I fixed this switching raspbian 64bits for raspbian lite 32bits

Can confirm that raspbian lite 64bit will cause similar issues for kiauh v6 install, raspbian lite 32bit installs smoothly