dw-0/kiauh #607: KConfig Support for Updates via KIAUH

smbenson0712, 2024-12-10:
### Is your feature request related to a problem? Please describe

No problem, just a convenience for multiple mcu's with different menuconfig settings.  

### Describe the solution you'd like

As many printers now are using multiple mcu's, it would be very nice to utilize KConfig via KIAUH.  This would make updating easier when you have to compile 2-3 mcu firmwares with different menuconfig settings.

### Describe alternatives you've considered

I usually update all the software with KIAUH and then use the command line to compile and flash the firmware of my mcu's using KConfig manually.

CODeRUS, 2025-01-17:
@smbenson0712 is that same with instructions provided by FYSETC for NIS?
https://github.com/FYSETC/Nozzle-Input-Shaper

I just encountered this problem myself.

@dw-0 what do you think about this, asking for a name for the file and then outputting a selection of .config files to build would be a sufficient solution?

dw-0, 2025-01-18:
> [@dw-0](https://github.com/dw-0) what do you think about this, asking for a name for the file and then outputting a selection of .config files to build would be a sufficient solution?

@CODeRUS  Im not 100% sure i understand your suggestion but maybe our ideas kind of match already?
I had some very rough ideas in the past how such a feature could be designed.
Lets say the user selected the following: `4) [Advanced] -> 1) [Build]`
- If no or only the default .config file exists, there could be a dialog at the end of the build process if the config should be saved as a custom config file.
- If there are already available .config files, the user can select one of the files and the build process would start with that config
  - There could be an option to create a new config even if there are already available files to create a new config from scratch
  - There could also be an option to delete config files (maybe part of a second iteration of that feature)

Is it a good idea to save those configs in their dedicated directory? This may make checks for existing files easier?

smbenson0712, 2025-01-18:
In my opinion I think that would be a perfect solution.  Easy to follow like all the other options in kiauh.

CODeRUS, 2025-01-19:
@dw-0 your explanation is totally correct

PR #628 (2025-02-09): Merge develop into master (Release v6.0.0-alpha.14)
fixes #607 
fixes #615
fixes #618 
fixes #619 
fixes #620 
fixes #623 
fixes #627
commit 0becf9d5 (2025-02-09): Release v6.0.0-alpha.14

fixes #607
fixes #615
fixes #618
fixes #619
fixes #620
fixes #623
fixes #627
closed 2025-02-09 by dw-0