dw-0/kiauh #632: Error parsing moonraker dependencies in V6

aymericblok, 2025-02-14:
### Linux Distribution

Raspberry pi os lite 32bit

### What happened

i installed klipper via V6, and then tried moonraker, but that wouldn't install. So i quit, restarted and tried in older version, and yes it works! see attached picture

### What did you expect to happen

Moonraker been installed like in the older version of kiauh

### How to reproduce

Raspberry Pi 4 Model B - 4 GB RAM -> fresh install of Raspberry pi os lite 32 bit from 2024/11/19 -> ran `sudo apt-get update && sudo apt-get install git -y` then ran -> `cd ~ && git clone https://github.com/dw-0/kiauh.git` and then ran -> `./kiauh/kiauh.sh` -> choose new version V6 -> installed klipper -> install moonraker -> see screenshot

### Additional information

[image]

huck0031, 2025-02-14:
Got the same issue when trying to install Moonraker, Raspberry Pi 3 model B running RaspiOS lite bookworm (2024/11/19). I have two instances of klipper installed

[image]

dw-0, 2025-02-15:
Hi! Thanks for the report. I was able to reproduce it and adapted the logic for reading the distro info before parsing the dependencies. Please try again and see if it works now.

closed 2025-02-15 by dw-0

aymericblok, 2025-02-15:
> Hi! Thanks for the report. I was able to reproduce it and adapted the logic for reading the distro info before parsing the dependencies. Please try again and see if it works now.

YES! this works perfectly! Thank you. 

Also, awesome to see your program updates so easily with the changes your made today, it was so easy to test your changes :). Thanks for your dedication!

PR #633 (2025-02-15): fix(moonraker): use os-release file to get distro info
fixes #632
PR #634 (2025-02-15): Merge develop into master (Release v6.0.0-alpha.15)
fixes #632
commit 1cf81377 (2025-02-15): Release v6.0.0-alpha.14

fixes #632
commit e3a0a9de (2025-02-15): Release v6.0.0-alpha.15

fixes #632