dw-0/kiauh #636: Install without sudo

keab, 2025-02-15:
### Is your feature request related to a problem? Please describe

I have tried to install on a Magisk-rooted Samsung S9 with linuxdeploy Debian, and my issue is that it's not possible to run commands with sudo - there is some security function killing every such attempt. I **can** SSH in as root and do any packet installation, but kiauh doesn't allow installation as root. 

### Describe the solution you'd like

I understand this is a very specific situation, but **if** there is a simple way to split the installation into a root part and a non-sudo part, that would have been useful. 

### Describe alternatives you've considered

The obvious and ideal way is to get a more 'complete' rooting of the device, but it's not my expertise and I have not been successful

dw-0, 2025-02-15:
> I understand this is a very specific situation, but if there is a simple way to split the installation into a root part and a non-sudo part, that would have been useful.

There is no simple way of doing this right now, as the whole logic is targeting non-root execution and using sudo where sudo is required. It would require a rework of a lot of different places i am currently not planning. Also it is very hard to reliably support niche use cases like that.

You could of course modify kiauh yourself, and remove the root check at the start of the application. Then remove all sudo keywords from the commands that will be executed during the actions you try to execute.

keab, 2025-02-15:
Thanks for responding! 
I assume the root check is there for a reason. What do I risk if following your suggestion?

lorenzzosd, 2025-03-06:
I'm having the same issue with my Galaxy S20 FE, and I'm also wanting to learn more about the risks of running as root.