dw-0/kiauh #639: kiauh Klipper Install error

weed2all, 2025-02-16:
### Linux Distribution

raspberry pi os lite (64 bit)

### What happened

try to install klipper using kiauh v6 and i get  Error cloning repository https://github.com/Klipper3d/klipper: Unknown error

[image]

### What did you expect to happen

install klipper

### How to reproduce

im using raspberry pi imager 1.8.5

dw-0, 2025-02-16:
This is usually some kind of network issue. It is not issue with KIAUH. Closing, as i cannot change anything in that situation.

closed 2025-02-16 by dw-0

weed2all, 2025-02-16:
i don`t have any network issues

weed2all, 2025-02-16:
v5 install just fine...network issues? go sleep brother...