dw-0/kiauh #651: ustreamer required pkg-config

galagithub, 2025-02-23:
### Linux Distribution

   _             _    _           
   /_\  _ _ _ __ | |__(_)__ _ _ _  
  / _ \| '_| '  \| '_ \ / _` | ' \ 
 /_/ \_\_| |_|_|_|_.__/_\__,_|_||_|
                                   
 v25.2.2 for Raspberry Pi 3 Model B Plus running Armbian Linux 6.12.15-current-bcm2711

 Packages:     Debian stable (bookworm)

### What happened

when installing crowsnest under armbian the script will eventually install ustreamer which in turn will search for pkg-config which is not installed by default on armbian.

please install pkg-config before attempting to install crowsnest and thus ustreamer

### What did you expect to happen

crowsnest success install

### How to reproduce

install crowsnest on a fresh armbian image

mryel00, 2025-02-26:
Can you send the output of the problem?

galagithub, 2025-02-26:
> Can you send the output of the problem?

I no longer can, I've upgraded that device.

but you can download armbian minimail and and install crowsnest using kiauh on a non-raspberry board.

mryel00, 2025-02-26:
That's my problem I currently don't have any non raspberry Pi boards, to check it 😅
I'm gonna look into ustreamer to find the dependencies, if they changed something. I updated the ustreamer version a while ago and only tested Raspberry Pi builds, since then.
Overall this is not a Kiauh issue but a Crowsnest issue. KIAUH is only installing what Crowsnest is telling it to install, afaik.

mryel00, 2025-02-26:
The only thing regarding `pkg-config` in ustreamer, was this comment: https://github.com/pikvm/ustreamer/issues/286#issuecomment-2294784430
Was the output similar?

I will try to reproduce it in a VM with Ubuntu the upcoming days, as @Ale11Re mentioned.

galagithub, 2025-02-26:
> The only thing regarding `pkg-config` in ustreamer, was this comment: [pikvm/ustreamer#286 (comment)](https://github.com/pikvm/ustreamer/issues/286#issuecomment-2294784430) Was the output similar?
> 
> I will try to reproduce it in a VM with Ubuntu the upcoming days, as [@Ale11Re](https://github.com/Ale11Re) mentioned.

Yes, it was just like that. It complains about not finding pkg-conf.

WayneManion, 2025-02-27:
I also encountered this problem on a BTT CB2 with BTT's (Armbian-based) beta OS release.

mryel00, 2025-02-27:
Will be fixed with https://github.com/mainsail-crew/crowsnest/pull/294
It's a new dependency that wasn't added to the Readme yet. It's not actually required for every installation, but it got added as some kind of safety measure, as you can read here: https://github.com/pikvm/ustreamer/issues/286#issuecomment-2689030128

mryel00, 2025-02-27:
PR is merged now and this issue should be fixed now

mryel00, 2025-02-27:
It was an update to Crowsnest, not kiauh, but thanks for confirmation

galagithub, 2025-02-27:
guess you can close this bug for now

mryel00, 2025-02-27:
Only the OP of the issue or a maintainer of a Github repo can close the issue.
So @galagithub I cannot close it, but you can

galagithub, 2025-02-27:
ok, seems that this issue has been fixed upstream, closing it.

closed 2025-02-27 by galagithub